Klipper3d/klipper #1586: Inverntech D1 (based on ReliaBot Delta V1 cofig file

peterlpop, 2019-04-27:
Not sure where to post this, but it took me several days of reading examples and tweaking this to get this printer going. Here's my config file - I'll move this somewhere or one of you guys can or just erase it.  Maybe this will save someone a lot of trouble (the hot end fan in your sample used a different pin than on my printer). 

#for printer Inverntech D1, a Reliabot Delta V1 printer (230x230x350)

[heater_fan my_nozzle_fan]
pin: ar7
#max_power: 
#shutdown_speed:
#cycle_time:
#hardware_pwm:
kick_start_time:0.5
heater: extruder
heater_temp: 40.0
fan_speed: 1.0

[stepper_a]
step_pin: ar54
dir_pin: !ar55
enable_pin: !ar38
step_distance: .01255
endstop_pin: ^ar2
homing_speed: 100
#position_endstop: 350.0
#arm_length: 264.0

[stepper_b]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .01255
endstop_pin: ^ar15

[stepper_c]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .01255
endstop_pin: ^ar19

[printer]
kinematics: delta
max_velocity: 300
max_accel: 3000
max_z_velocity: 150
#delta_radius: 174.75
minimum_z_position: -20

[fan]
pin: ar9

#[endstop_phase]

[extruder]
step_pin: ar26
dir_pin: !ar28
enable_pin: !ar24
step_distance: 0.01043
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250
max_extrude_only_distance: 1000.0

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control: watermark
#pid_Kp: 22.2
#pid_Ki: 1.08
#pid_Kd: 114
min_temp: 0
max_temp: 130

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
pin_map: arduino

[probe]
z_offset: -0.08
pin: ^!ar18

[delta_calibrate]
radius: 80
speed: 50
horizontal_move_z: 5
samples: 5
samples_result: average
sample_retract_dist: 1.0

[display]
lcd_type: hd44780
rs_pin: ar16
e_pin: ar17
d4_pin: ar23
d5_pin: ar25
d6_pin: ar27
d7_pin: ar29
encoder_pins: ^ar31, ^ar33
click_pin: ^!ar35
kill_pin: ^!ar41

[gcode_macro G29]
gcode:

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [printer]
#*# delta_radius = 134.093211
#*#
#*# [stepper_a]
#*# angle = 209.424082
#*# arm_length = 264.000000
#*# position_endstop = 350.002654
#*#
#*# [stepper_b]
#*# angle = 328.837169
#*# arm_length = 264.000000
#*# position_endstop = 332.439432
#*#
#*# [stepper_c]
#*# angle = 90.000000
#*# arm_length = 264.000000
#*# position_endstop = 348.380033
#*#
#*# [delta_calibrate]
#*# height0 = -0.08
#*# height0_pos = 27870.000,26510.000,27775.000
#*# height1 = -0.08
#*# height1_pos = 30919.000,29610.000,25366.000
#*# height2 = -0.08
#*# height2_pos = 27098.000,31483.000,27003.000
#*# height3 = -0.08
#*# height3_pos = 25634.400,29072.400,30381.400
#*# height4 = -0.08
#*# height4_pos = 27128.600,25739.600,31310.600
#*# height5 = -0.08
#*# height5_pos = 30037.000,24435.000,29942.000
#*# height6 = -0.08
#*# height6_pos = 32086.000,25773.000,27003.000

KevinOConnor, 2019-04-28:
We can add an example config file.  Can you provide a link to the manufacturer's page for the printer and can you provide the year (roughly) the printer you have is from?

-Kevin

peterlpop, 2019-04-28:
reliabot.com clone - iverntech.com is gone - 2019? for iverntech D1?

It's an Atmega 2560 based on reliabot.com v1.3- though as of now, it looks like I might have bought the last one on Amazon (ivenrntech.com is gone) and while it worked really good, the bed heater won't turn on anymore (and it's the board that won't do it). Not sure when reliabot.com came out with their board.

KevinOConnor, 2019-05-25:
Thanks.  I'm going to close this issue - hopefully those with the printer will still be able to find the attached config when doing a search.

-Kevin

closed 2019-05-25 by KevinOConnor

alessiodueb, 2019-11-17:
Hi sorry for the question, is this configuration file for Marlin firmware? I'm desperate, 
I can't find the motherboard of Reliabot Printer in the marlin board section, sorry can you help me?

thanks in advance for your reply

peterlpop, 2019-11-17:
@alessiodueb - you are on Klipper firmware forum - I am sure others can describe it better - but this Firmware runs on a Raspberry PI connected to your printer, effectively moving the Marlin firmware all processing from the Atmega 2560 to the much faster rPI. 

If you are looking for Marlin firmware and config files for the main board, you are in the wrong spot. - the configuration file above is for the raspberry-pi-ran firmware of Klipper.

alessiodueb, 2019-11-17:
Hi, thank you for the reply, i know that's the wrong Place, but i'm
desperate and i've tryed, thanks anyway for the information

peterlpop, 2019-11-18:
well, you might be in the right place still. Get a raspberry pi, load Klipper on your printer's board and you'll have far more control and easier config (that config file) - and way faster and better prints than with any other printer hardware or firmware I have tried. Klipper rocks. And all that seamless OctoPrint and Cura.

alessiodueb, 2019-11-18:
Thank you, it's the best news you could give me, i Know Raspberry and i use
it for a lot of project so i'm in my comfort zone,it may be easier to use
Raspberry than the original motherboar,I will definitely try, thanks again

harrylarry00, 2020-10-10:
Hey quick question I have the same printer but when I put it together the lcd just gives me some boxes on the screen and that’s about it. Do I have to install software or something?
[image]

alessiodueb, 2020-10-10:
hi, I can't see the image you sent me but as I remember, together with the
printer they gave you an SD memory in which there are the firmware to be
flashed inside the printer controller, the instructions should be in the
README.txt file

harrylarry00, 2020-10-10:
Got it I’ll try it out tomorrow and let you know how it goes👍🏼

alessiodueb, 2020-10-10:
If it suits you I have the originals they gave me with the printer,
tell me if you have solved it 😉

harrylarry00, 2020-10-22:
Sorry I’ve been busy so I haven’t had time to fix it but I was able to load the software but it’s there a setting I forgot to enable because the nozzle stays at 0 and doesn’t even try to preheat and the bed is stuck at 25 but says it’s going to heat up to 500f which I know is damn well off.

harrylarry00, 2020-10-22:
Is*

alessiodueb, 2020-10-22:
have you checked the electrical connections?  In case everything is fine,
if you check the settings of the Cura slicer you should be able to change
the temperature before loading the program