Klipper3d/klipper #1588: Anycubic Delta - TMC2208 - TriGorilla Unable to read tmc2208 'extruder' register IFCNT

totalitarian, 2019-04-28:
I have the following TMC2208 drivers and have bridged the pins as show

[image]

I then hooked up to my TriGorilla board on pin D11

[image]

Here is my config

[tmc2208 extruder]
uart_pin = ar11
microsteps = 32
run_current = 0.8
hold_current = 0.4
stealthchop_threshold = 200

This is the error I get

Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 137, in _connect
    cb()
  File "/home/pi/klipper/klippy/extras/tmc2208.py", line 327, in _init_registers
    self.set_register(reg_name, val)
  File "/home/pi/klipper/klippy/extras/tmc2208.py", line 348, in set_register
    self.ifcnt = ifcnt = self.get_register("IFCNT")
  File "/home/pi/klipper/klippy/extras/tmc2208.py", line 340, in get_register
    "Unable to read tmc2208 '%s' register %s" % (self.name, reg_name))
Error: Unable to read tmc2208 'extruder' register IFCNT
Unable to read tmc2208 'extruder' register IFCNT
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

[klippy.txt](https://github.com/KevinOConnor/klipper/files/3124754/klippy.txt)

Hywelmartin, 2019-04-30:
have you tried another pin. Like ar5 .. a while ago some one claimed that the servo pins where in reverse... #1448

totalitarian, 2019-05-01:
Afraid I tried 4,5,6,11,12 and none worked

jakep82, 2019-05-01:
That indicates a likely issue with the driver itself.

totalitarian, 2019-05-01:
I have tried all 4 drivers, you are right it could be the drivers are not wired correctly from the factory.

james-golden, 2019-05-03:
Did you bridge all three solder pads?

totalitarian, 2019-05-03:
> Did you bridge all three solder pads?

Yes, on two set I bridged 2 pads and on the other 2 sets I bridged 3 pads. Confirmed continuity to the pins to make sure my soldering was correct

james-golden, 2019-05-04:
The 3 solder pads sets the pin the uart is on. Check the data sheet. It will help bunches.  You might also need a resistor on the pin. Data sheet will tell you much more

james-golden, 2019-05-04:
https://www.trinamic.com/fileadmin/assets/Products/Eval_Drawings/SilentStepStick-TMC2208_v11.pdf

totalitarian, 2019-05-04:
@james-golden The pads are soldered and the resistor is not needed with klipper

Paulbsh, 2019-05-07:
Which pin of the driver did you connect? Pics from the board with the driver on it?

aproemel, 2019-05-12:
I have similar issues with my I3 Mega with trigorilla board.

Under Marlin my TMC2208 work, I've even tested to limit the current and use fast moves to check for a proper communication. 
Result: 
- with low current, fast movements not possible
- withg 800mA current, fast movement was possible

I'm using HW Serial1, 2 and 3 as also 2 SW Serial ports (Marlin bugfix 2.0 branch)
Part from my marlin config:
  #define X_HARDWARE_SERIAL  Serial1
  #define Y_HARDWARE_SERIAL  Serial3
  #define Z2_HARDWARE_SERIAL Serial2
  #define Z_SERIAL_TX_PIN    5			//SERVO01
  #define Z_SERIAL_RX_PIN    12		
  #define E0_SERIAL_TX_PIN   6			//SERVO02
  #define E0_SERIAL_RX_PIN   11			//SERVO00

part from my klipper config:
[tmc2208 stepper_x]
##Serial1
uart_pin: ar19
#tx_pin: PD3
microsteps: 16
run_current: 0.800
sense_resistor: 0.110
stealthchop_threshold: 60

I've tested PD2, ar19 for RXD1 which does not influence the result.
I've tested with and without tx_pin, also with not influence to the result.
I've tested all 5 TMC2208, I would have 4 more on spare, but I don't think it's driver related as Marlin communicates with them and my test result even makes me more confident in beleving so.

> Extruder max_extrude_ratio=0.266081
> Starting serial connect
> got {'#receive_time': 1277.047949617, u'next_clock': 3259665408L, u'oid': 20, u'value': 7856, '#name': u'analog_in_state', '#sent_time': 1276.962066231}
> got {'#receive_time': 1277.217947273, u'next_clock': 3262385408L, u'oid': 7, u'value': 7856, '#name': u'analog_in_state', '#sent_time': 1277.2002801370002}
> got {u'count': 148, '#receive_time': 1277.2531754499998, u'sum': 304811, u'sumsq': 3427992, '#name': u'stats', '#sent_time': 1277.251391023}
> got {'#receive_time': 1277.3486997730001, u'next_clock': 3264465408L, u'oid': 20, u'value': 7854, '#name': u'analog_in_state', '#sent_time': 1277.302624564}
> Loaded MCU 'mcu' 77 commands (v0.7.0-457-g777013d-20190510_191316-octopi / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
> MCU 'mcu' config: RECEIVE_WINDOW=192 BUS_PINS_spi=PB3,PB2,PB1 SOFT_PWM_MAX=256 SERIAL_BAUD=250000 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 RESERVE_PINS_serial=PE0,PE1 BUS_PINS_twi=PD0,PD1 MCU=atmega2560 STATS_SUMSQ_BASE=256 CLOCK_FREQ=16000000
> got {'#receive_time': 1277.518695554, u'next_clock': 3267185408L, u'oid': 7, u'value': 7856, '#name': u'analog_in_state', '#sent_time': 1277.5098245640002}
> Configured MCU 'CLOCK_FREQ' (611 moves)
> Config error
> Traceback (most recent call last):
>   File "/home/pi/klipper/klippy/klippy.py", line 137, in _connect
>     cb()
>   File "/home/pi/klipper/klippy/extras/tmc2208.py", line 327, in _init_registers
>     self.set_register(reg_name, val)
>   File "/home/pi/klipper/klippy/extras/tmc2208.py", line 348, in set_register
>     self.ifcnt = ifcnt = self.get_register("IFCNT")
>   File "/home/pi/klipper/klippy/extras/tmc2208.py", line 340, in get_register
>     "Unable to read tmc2208 '%s' register %s" % (self.name, reg_name))
> Error: Unable to read tmc2208 'stepper_x' register IFCNT

Is there any possibility to debug the root cause?

Thanks in Advance
BR
Alex

KevinOConnor, 2019-05-25:
I'm going to close this issue as it looks like the conversation has concluded.  To the best of my knowledge, the Klipper tmc2208 uart code does work - getting a connection to the uart pin can be tricky however on some stepsticks.

-Kevin

closed 2019-05-25 by KevinOConnor

lych, 2019-09-26:
TriGorilla doesn't have microstep jumpers so to avoid connecting the uart pin to VCC one have to cut or bent this pin on TMC2208 (probably it's better to do the same thing with MS1 and MS2 pins).
Also, I don't know if it's a problem in Klipper or in the TriGorilla documentation, but for Klipper right order of servo pins looks like this : D11 D4 D6 D5 D12.
[image]

Hywelmartin, 2019-09-27:
As you have v1.1 of that board the pinout is as mentioned.. the one without versioning got 4,5,6,11,2