Klipper3d/klipper #1589: Problem with temperature stability

JSGUYOT, 2019-04-28:
Hi !

I have a problem with tempereture extruder sensor.
Below, the capture of "PID_CALIBRATE HEATER=extruder TARGET=170".

[image]

The temperature rises too high before descending.
I test with another sensor, same issue ...
I use a NTC3950 with this paramters, on board MKS Gen 1.4 :

> [extruder]
> step_pin: ar36
> dir_pin: ar34
> enable_pin: !ar30
> step_distance: .0022
> nozzle_diameter: 0.400
> filament_diameter: 1.750
> heater_pin: ar7
> sensor_type: NTC 100K beta 3950
> sensor_pin: analog13
> control: pid
> pid_Kp: 18.877
> pid_Ki: 0.707
> pid_Kd: 126.002
> min_temp: 0
> max_temp: 250

I haven't any more idea of the root cause ... can you help me ?

In attachment, klippy.log file.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3124808/klippy.log)

Regards,

thucar, 2019-04-29:
It's ok if the temperature fluctuates like that during calibration. If you set the target to 170C after calibration, does it still fluctuate as much?

JSGUYOT, 2019-04-29:
Hi ! 

Yes it's OK when i set to 170°C without PID calibration, the temperature is stable. 
I should have tested this before opening this issue, but I was stuck on this point.
Sorry for my misunderstanding for this point ...

Regards,

closed 2019-04-29 by JSGUYOT