Klipper3d/klipper #1591: Config file for Ender 5?

VanceK, 2019-04-28:
Any chance of providing a config file for the Creality Ender 5?  I'd like to utilize Klipper but didn't have any luck getting the Z-Axis to play nice - got homing to work but Z movements are reversed - so I went back to Marlin for now.  I don't have a log file to post at the moment. Thanks for any help!

https://www.creality3d.cn/creality-ender-5-p00261p1.html

xelprep, 2019-04-29:
I hope I did this right, but in case it doesn't get merged for some reason feel free to check out PR #1594 for a config I've been using with my Ender 5. I also use the following config for this printer for manual mesh bed leveling:

```
[bed_mesh]
speed: 50
horizontal_move_z: 5
samples: 3
samples_result: average
sample_retract_dist: 2.0
min_point: 20,20
max_point: 215,215
probe_count: 3,3
```

But I didn't include it in the example config because the Ender 3 didn't include any additional configs. I'm new to 3D printing and Klipper so it might not be perfect, but it's been working for me so far.

VanceK, 2019-04-29:
Thank you, @xelprep! Your PR file did the trick!

KevinOConnor, 2019-05-01:
FYI, this was commited to the main repo (f444177b).

-Kevin

closed 2019-05-16 by KevinOConnor