Klipper3d/klipper #1593: Bed tilt calibrate - 3 Z steppers or 4?

HoshuaK, 2019-04-28:
Hi there. Not really an issue but can't find anywhere what will work best for klipper. I'm building a new printer soon and plan of having a multiple Z stepper moving bed and can't find any information on whether to build it with 3 steppers or 4 one at each corner. Any help would really be appreciated. Also if there is a better place to ask these sorts or questions that would be great too.

KevinOConnor, 2019-05-01:
That's a pretty wide open question, so I suspect it will be difficult to get a conclusive answer.  FWIW, you can research the Railcore ZL (which uses 3 z steppers) and the Voron2 (which uses 4 z steppers).  There may be more info there.

-Kevin

sniperlucian, 2019-05-06:
since I am already logged in:

I use three stepper for Z.  Simple reason:  4 steppers are over constrained system and might fight each other.  (Best example is a three leg stool - it never wobbles.)

HoshuaK, 2019-05-06:
Thank you. I was unsure if either 4 or 3 were supported. Looks like both but I'll use 3. Thanks.

closed 2019-05-06 by HoshuaK