Klipper3d/klipper #1595: Mesh extends outside of the fade range

CorvetteCole, 2019-04-29:
So, while using a 3x3 grid for probing, when I try to use a 20x20 grid or anything around that size I get this error:
Mesh extends outside of the fade range, please see the fade_start and fade_end options inexample-extras.cfg. fade distance: 8.00 mesh min: -7992.4127mesh max: 7226.9830.

I don't really understand what I am missing here regarding this. I suspect this is my error but I cannot figure out why.

Thanks in advance, my log is attached below.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3125943/klippy.log)

Arksine, 2019-04-29:
Too many probe points for the lagrange algorithm.  Above 7 or so will result in oscillation, creating huge peaks and valleys in the mesh when attempting to interpolate.

When probing that many points you have two options:
1) set the `algorithm` option to bicubic
2) set `mesh_x_pps` and `mesh_y_pps` to 0.  This will disable interpolation (mesh interpolation isn't really necessary when probing so many points)

CorvetteCole, 2019-04-29:
brilliant, thanks for the response. That is somewhat close to what I figured but I had no idea. Which of the options do you think would be better? I do not know the difference between bicubic and Lagrange and Google is not helping much. Thanks again for the response

Arksine, 2019-04-30:
On a standard size printer (ie 200mm x 200mm) I would set mesh_x_pps and mesh_y_pps to 0 if sampling a 20x20 grid.  I don't think you gain anything by interpolating extra points in the mesh with a grid that dense.   You can leave the algorithm on lagrange, bed_mesh detects when the user specifies zero points to interpolate and disables it.

CorvetteCole, 2019-04-30:
Alright, sounds good. Thanks again for your help

closed 2019-04-30 by CorvetteCole