Klipper3d/klipper #1596: Getting Failed to home x timeout during endstop homing

iamthebest22, 2019-04-29:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3126014/klippy.log)
I'm trying out Klipper, and did all the config checks, the stepper buzz check also works perfectly, but when I input G28 or G28 X or G28 Y to try and home, it gives the same Failed to home x or home y : timeout during endstop homing. At first I thought it was issue #196, cause I was using TMC2208's, so I switched to LV829's with 1/8 microstepping and steps distance set to 0.02, but still the same error, here is my printer.cfg too 

https://www.dropbox.com/s/yl4ck275spmnbns/printer.cfg?dl=0

the error happens around 1 second after I send the G28 commands. 
It's a test bench that I created, so no actual printer, I would be pressing the endstop to manually trigger, but the motors would not even move after the G28 command and as said above would just give the error. Thank you very much for the help.

iamthebest22, 2019-04-30:
I just found the problem so nvm.

closed 2019-04-30 by iamthebest22

jerkwagon, 2020-11-01:
cool whatever you do, do share the solution because you know..

ratwalker, 2021-04-11:
HI,BRO. I guess some motor Pulse/Rev maybe too high,check the all motor microstep ,like your ([tmc2130 stepper_x] microsteps = 256).Sincerely.