Klipper3d/klipper #1597: Psu control

dugly66, 2019-04-29:
I have klipper working perfectly at this point but I have been trying to get the psu control back. I have read through the related issues but none of them seem to actually explain how to set it up. Am I missing something somewhere?

dugly66, 2019-04-29:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3128751/klippy.log)

cmock, 2019-04-30:
You already have defined the pin, so to use it just say `SET_PIN PIN=power VALUE=1`; you could also define gcode macros for M80/M81 to invoke that.

dugly66, 2019-04-30:
Thank you. I can control it now be  sending that in the octoprint terminal and make it turn on. How do I re-automate that so it does it like before with the m80 m81 gcodes?

KevinOConnor, 2019-05-16:
I'm going to close this issue as it looks like a solution to the original issue was found.

-Kevin

closed 2019-05-16 by KevinOConnor