Klipper3d/klipper #1598: Looking for configuration overview

VanKurt, 2019-04-30:
I've read through all (?) of the documents in the repository. But I can't find acomplete overview of all configuration options anywhere (e.g. for TMC2208 stepper drivers, blTouch etc.).

Am I just blind? Or is such an overview missing from the docs?

theopensourcerer, 2019-04-30:
Have you looked through the many config example files... e.g.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L1018

VanKurt, 2019-04-30:
> Have you looked through the many config example files... e.g.

I have. But since this project is so awesomely documented in every other aspect, I refused to believe that there is no configuration overview (which is maybe the most important part of getting Klipper to run) ;-)

So I will dig through all the examples and puzzle together what I need. Luckily the configuration seems to be very straight forward.

closed 2019-04-30 by VanKurt