Klipper3d/klipper #1599: [Request] New restart_method GPIO (RPi to AVR direct serial connection)

konkav-js, 2019-04-30:
I have a printer with ATmega2560 (Anycubic 4MAX). I bought an RPi 3 A+, because there is enough free space beside the board to put the RPi inside the enclosure. The board (Trigorilla) has the UART3 on headers, and has a nice, capable DC-DC converter (it could power the RPi as well). So I installed the OctoPi and Klipper on the RPi, set up the RPi serial (disabled the console), moved an endstop to other pin, and got it working (yes, connected a logic lever converter). I have printed few prints, it is working nice. But. The Klipper can't restart the m2560. If I connect them by USB, everything works great, but I want to get rid of the USB cable.
RPi has a bunch of GPIO pins, and I though, that it would be an easy fix, to have one GPIO connected through a transistor to the reset pin of the m2560 (it is available on the ISP header). What do you think about it?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3133080/klippy.log)

jakep82, 2019-04-30:
I'm surprised 'restart_method = command' isn't working.  I have a sam3x8e MCU connected via UART to a Pi 3B, and it works as expected for me using command.

konkav-js, 2019-05-01:
Unfortunately not. Now I am powering the RPi from a phone charger, to have the ability to power cycle the printer, if there is a need for restart. I hope that this GPIO restart is not too complicated to code (unfortunately I am not a programmer).

KevinOConnor, 2019-05-02:
The "restart=command" mechanism is the preferred method for implementing the above.  If that's not working for you, please attach a Klipper log file with a session that includes a FIRMWARE_RESTART command.

It's possible to control the rpi gpio from Klipper.  To do that would require porting the linux micro-controller code to the rpi and adding gpio support to it.  That's a bit of software development work.  If someone wishes to do that and submit it, that's great.

-Kevin

konkav-js, 2019-05-03:
OK, I have done a test for the log. Stopped the Klipper, deleted the log, started the Klipper, started the printer (Mega2560), issued the FIRMWARE_RESTART in Octoprint Terminal. Klipper restarted, but the Mega just frozen. I issued a STATUS twice, got Printer no ready. Then power-cycled the printer. After that, the printer connected to Klipper. I issued a plain RESTART, and the connection remained.

Raspberry Pi 3 A+ with OctoPi+Klipper

[klippy.log](https://github.com/KevinOConnor/klipper/files/3142788/klippy.log)

konkav-js, 2019-05-04:
Well, yesterday I had to try out something, so I searched around a bit. There is a Python library for easy manipulation of GPIO ports, and this lib is installed by default on Raspbian. But unfortunately it were not in Klipper's venv. I went to install it by PIP. The code for toggling the pin 18 (SoC pin number):
Init:
```python
import RPi.GPIO as GPIO
from time import sleep

GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.OUT, initial=GPIO.HIGH)
```
Toggle:
```python
    GPIO.output(18, 0)
    sleep(1)
    GPIO.output(18, 1)
```
It worked nice. But I thought that maybe this is not a good way, this library should be installed only on RPi systems, so I searched for more universal method. And there is. On Raspberrys there is a possibility to access a GPIO pin in `/sys/class/gpio/` . This is more usable, because on init, we could search for this path. If there is, then the method is usable. So I added to the `/etc/init.d/klipper` script:
To the `start)` case:
```bash
if [ -d "/sys/class/gpio/" ]; then
    echo "18" > /sys/class/gpio/export
    echo "out" > /sys/class/gpio/gpio18/direction
    echo 1 > /sys/class/gpio/gpio18/value
fi
```
And to the `stop)` case:
```bash
if [ -d "/sys/class/gpio/" ]; then
    echo "18" > /sys/class/gpio/unexport
fi
```
Then I edited `mcu.py`:
- added `import rpigpio`
- in `def __init__` extended to: `rmethods = {m: m for m in [None, 'arduino', 'command', 'rpi_usb' ,'rpi_gpio']}`
- under `class MCU:` 
  - changed to :
```python
    def microcontroller_restart(self):
        if self._restart_method == 'rpi_usb':
            self._restart_rpi_usb()
        elif self._restart_method == 'rpi_gpio':
            self._restart_rpi_gpio()
        elif self._restart_method == 'command':
            self._restart_via_command()
        else:
            self._restart_arduino() 
```
  - and added :
```python
    def _restart_rpi_gpio(self):
        logging.info("Attempting MCU '%s' reset via rpi GPIO toggle", self._name)
        self._disconnect()
        rpigpio.rpi_gpio_reset()
```
- deleted the `mcu.pyc` (not sure this is needed)

- in the same directory created `rpigpio.py` with content:

```python
from time import sleep
from subprocess import call

def rpi_gpio_reset():
    call(["echo 0 > /sys/class/gpio/gpio18/value"],shell=True)
    sleep(1)
    call(["echo 1 > /sys/class/gpio/gpio18/value"],shell=True)
```
And in the `printer.cfg` under `[mcu]` changed to:
`restart_method: rpi_gpio`

And, it's working 😄 This was a fun search. I have never figured out things in software this complex. I hope, maybe some programmer stops by, and feels that this would be easy to add, and adds this to Klipper. Unfortunately I feel that this is over me: in printer config, there should be a possibility to enter the pin number, and the pin init really shouldn't be in init.d script, but in Klipper init, and I have no idea, how and where to start.

At the HW level all somebody need to get it connected is at least 4 channel logic level converter, which can be had for few dollars from eBay. Two channels are needed for the serial, and one for reset. If this gets implemented, I would gladly write a manual on how to make the connections.

KevinOConnor, 2019-05-08:
The symptoms you describe sound similar to: https://www.klipper3d.org/FAQ.html#when-i-set-restart_methodcommand-my-avr-device-just-hangs-on-a-restart

There's some info on bootloaders at: https://www.klipper3d.org/Bootloaders.html

-Kevin

closed 2019-05-25 by KevinOConnor