Klipper3d/klipper #1600: Upgrade script Question

novio8, 2019-05-01:
I would like to upgrade to latest version. I run Klipper on Ubuntu. I'v used install-ubuntu-18.04.sh script for installation, should I use the same for upgrade ?
Are the following steps correct and if all my settings remain intact ?

1. I downloaded latest Zip from Github
2. Unpacked the file.
3. Run Script from unpacked folder.

Regards
Mario

KevinOConnor, 2019-05-01:
The recommended upgrade procedure is described at: https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software 

You'll need to adapt those instructions to your custom Ubuntu setup.

-Kevin

novio8, 2019-05-02:
Kevin thank you,
are left intact when upgrading following procedure in FAQ
regrds
MArio

MoshiBin, 2019-05-04:
Procedure in FAQ doesn't alter `~/printer.cfg` - not even `install-octopi.sh`.

novio8, 2019-05-05:
MoshiBin thank you

closed 2019-05-05 by novio8