Klipper3d/klipper #1601: [Recommendation request] Suggestions on how to associate accelerometer data with carriage movement

w1ebr, 2019-05-01:
Hello Kevin,

My printer has an accelerometer sensor attached to an extruder carriage. It is wired directly to the Raspberry Pi via I2C. I have sample python code that can configure the sensor and read the x and y axis acceleration data.

I want to be able to associated the data with the movement of the carriage. Initially I will configure the sensor to provide about 50 samples/sec. The samples have to be read via I2c as an x,y pair every 50th of a second

There is a version with a small FIFO which I could purchase which would reduce the bus read frequency by about 4x. Also, if connecting the accelerometer to the MCU instead of to the Raspberry Pi would make associating the sensor data with the carriage movements easier, I can buy a second sensor and wire it up to the MCU.

What code and/or documentation should I look at as a starting point to writing my own routine?

Thank you, Kevin!

Gene

w1ebr, 2019-05-04:
Also, is there a way to plot from Python 2.7? It seems that mathplotlib requires python 3.5 or higher
Thanks!

KevinOConnor, 2019-05-16:
Alas, I don't have any suggestions for wiring an accelerometer to Klipper.  There's been some discussion of using a phone accelerometer for testing purposes at the end of issue #57.

If you do come up with something, let us know.

-Kevin

w1ebr, 2019-05-16:
Hi Kevin,
Thanks for responding. I have a lot more homework to do. When I have something worth discussing, I''ll post it here.
Gene

KevinOConnor, 2019-05-25:
Thanks.  I'm going to close this issue as it looks like the conversation has concluded.

-Kevin

closed 2019-05-25 by KevinOConnor