Klipper3d/klipper #1603: Error: Requested temperature (2050.0) out of range

jstreit15, 2019-05-02:
Trying to print and it gets ready to go and then it thinks that temp of 2050c has been requested.  has happened twice now.

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/3136701/klippy.log)

jstreit15, 2019-05-02:
[klippy2.log](https://github.com/KevinOConnor/klipper/files/3136715/klippy2.log)
i noticed in the uploaded one it really truncated it... trying again.

jakep82, 2019-05-02:
Have you inspected the gcode you're trying to print?

jstreit15, 2019-05-02:
I will take a look tonight... its a very simple 40mm x 40mm x 1mm square that I am trying to do a test print with so the code shouldn't be hard :)  now that you say that... my temp was set to 205 which i guess 2050 isn't hard to do if you forget a period... maybe I found a flaw in cura 4.0

jstreit15, 2019-05-02:
looks like it was a fat finger situation in cura.  I had print temp set to 2050 instead of 205.0.  Sorry about that.  going to go ahead and close this ticket.

closed 2019-05-02 by jstreit15