Klipper3d/klipper #1604: [Request] User-defined menu items

alfsoft, 2019-05-03:
I am unable to find any docs explaining how to add my own items to Klipper menu that I defined in my printer.cfg. Or is this feature simply missing?

The point is that I thought that all my menu items that I defined in printer.cfg like described in example-menu.cfg would just appear in a new Klipper sub-menu. Something like "User-defined". But no, Klipper menu stays intact with all menu items definitions in printer.cfg.

Here's what I added to my cfg file:

```
[gcode_macro PARK_MACRO]
default_parameter_X: 10
default_parameter_Y: 10
default_parameter_Z: 10
gcode:
    M117 PARKING...
    G91
    G1 E-3.14 F1500
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F3000
   
[gcode_macro UNLOAD]
gcode:
    G91
    G1 E5.0 F1200
    G1 E3.0 F1600
    G1 E-13.14 F7000
    G1 E-80 F3000
    G90
       
[gcode_macro M600]
gcode:
    PAUSE_MACRO
    UNLOAD
 
[gcode_macro PAUSE_MACRO]
gcode:
    PAUSE
    PARK_MACRO
   
[gcode_macro PURGE]
gcode:
    M117 PURGING...
    G91
    G1 E45.0 F250
    G90
       
[gcode_macro LOAD_FILAMENT]
gcode:
    M117 LOADING...
    G91
    G1 E45.0 F250
    G4 P900
    G1 E45.0 F250
    G90
 
[gcode_macro RESUME_MACRO]
gcode:
    M117 RESUMING...
    RESUME

[menu command1]
type:command
name: Change Filament
gcode:
    M600

[menu command2]
type:command
name: Purge Filament
gcode:
    PURGE

[menu command3]
type:command
name: RESUME
gcode:
    RESUME_MACRO
```

Change Filament, Purge Filament and RESUME are my menu definitions that are still missing in menu. So altering klippy/extras/display/menu.cfg is the only way of adding such items for now?

AndyWo83, 2019-05-05:
Hi,
you can change it in "/home/pi/Klipper/klippy/extras/display/menu.cfg"
I have also made a lot of changes in my menu.cfg...

Regards Andy

alfsoft, 2019-05-05:
@AndyWo83 Yes, thank you. But won't it be overwritten during Klipper update?

AndyWo83, 2019-05-05:
@alsoft No, Klipper can not be update, when you make changes by yourself.
You must take a complete new install...
Or you save the old menu.cfg and when you will take a update from Klipper, replace your file with the old one....

danielfmo, 2019-05-05:
@AndyWo83 what you suggest is not recommended neither the way to go! 
To edit the menu you just need to override the items that you wan on your printer.cfg.

If you want to change all, copy all it's contents and edit as you like.

theopensourcerer, 2019-05-05:
@AndyWo83 What do you mean? Klipper can be easily updated. I do it all the time and it does NOT change your .cfg file(s).

Once klipper is installed, essentially, you only need to do a "git pull", make, make flash to update klipper...

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software

alfsoft, 2019-05-05:
Omg, now I get it. Now I see how it is supposed to work. Thank you all for your valuable help!
Added this text to my printer.cfg to fit all my needs for now:
```
[menu __main]
type: list
name: Main Menu
items:
    __tune
    __user
    __octoprint
    __sdcard
    __control
    __temp
    __filament
    __prepare
    __test

[menu __user]
type: list
name: USER
items:
    .__userfilchange
    .__userfilpurge
    .__userfilresume

[menu __user __userfilchange]
type:command
name: Change Filament
gcode:
    M600

[menu __user __userfilpurge]
type:command
name: Purge Filament
gcode:
    PURGE

[menu __user __userfilresume]
type:command
name: RESUME
gcode:
    RESUME_MACRO

```
This is a great way to add my own menu elements without editing menu.cfg and other system files.

closed 2019-05-05 by alfsoft