Klipper3d/klipper #1605: Feed Filament menu item is too long

alfsoft, 2019-05-03:
As you can see the Feed Filament menu item value is off-screen that makes this item useless:
[image]

Maybe we need to shorten this item's name a bit? "Feed Fil.:" or "Feed Flmt:" for example?

AndyWo83, 2019-05-05:
Hi,
you can change it in "/home/pi/Klipper/klippy/extras/display/menu.cfg"

OLD:
[menu __filament __feed]
type: input
name: Feed Filament: {0:.1f}
parameter: 0
input_step: 0.1
gcode:
    M83
    G1 E{0:.1f} F30

NEW:
[menu __filament __feed]
type: input
name: Feed Fil: {0:.1f}
parameter: 0
input_step: 0.1
gcode:
    M83
    G1 E{0:.1f} F30

Regards Andy

alfsoft, 2019-05-05:
@AndyWo83 Thanks. I made a new pull request: https://github.com/KevinOConnor/klipper/pull/1609

closed 2019-05-05 by alfsoft