Klipper3d/klipper #1607: Ender 3 LCD on SKR 1.3

banolka, 2019-05-03:
Hi I'm changing my Ender 3 to SKR 1.3 from BIGTREETECH with TMC2280 UART and I have a problem with LCD. 

Here is my display config:

```
[display]
lcd_type: st7920
cs_pin: P1.19
sclk_pin: P1.20
sid_pin: P1.18
encoder_pins: ^P3.26, ^P3.25
click_pin: ^!P0.28

[output_pin beeper]
pins: !P1.30
```
I try to connect lcd with 2 IDC cables but with no luck :( my ender has EXP1, EXP2 and EXP3) how should i connect this?

closed 2019-05-03 by banolka

harleycarr, 2019-05-21:
@banolka Did you get this working? We need cr10-stock display option for SKR V1.3.

povlhp, 2019-05-23:
On my original ender-3 board, I had this working
`[display]
lcd_type: st7920
cs_pin: PA3	#	- LCD_PINS_RS
sclk_pin: PA1	# - LCD_PINS_D4
sid_pin: PC1	- #  LCD_PINS_ENABLE 
encoder_pins: ^PD2, ^PD3	#BTN_EN2, BTN_EN1                                        
click_pin: ^!PC0	#- BTN_ENC 
`

Here we have an image of standard pinout
https://i.imgur.com/45OnGS5.jpg 
Somebody likely tried to translate that into the SKR 1.3 config in a wrong way.

To fix the beeper use as below. Not the inverted.
`[output_pin beeper]
pin: P1.30
`

I cant find original pinout, but they likely merged 2 connectors into one, using a non-standard layout.  From the Pin 27 BLTouch hack, we know the location of the beeper. We get light in display, so +5V and GND are likely fine as well.

The commented out section in SKR 1.3 config uses some P3.25/P3.26 for encoder. That is wrong as they are not on our pinout,

I assumed the MOSI / SID, CLK and CS was copied down from EXT2 to EXT in same order just after 5V. And that worked. The buttons was more difficult, tried a few of the possible options, and ended up with this, which works on my SKR v1.3 and ender-3 display:

`## RepRapDiscount Full Graphic Smart Controller ##

[display]
lcd_type: st7920
cs_pin: P1.19
sclk_pin: P1.21
sid_pin: P1.23
encoder_pins: ^P1.18, ^P1.20
click_pin: !P0.28
#
[output_pin beeper]
pin: P1.30
`