Klipper3d/klipper #1608: klipper timer to close

tonn0297, 2019-05-05:
hello i get a timer too close after 1-2 h of print. seems like its when theres a lot of small motions.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3145379/klippy.log)

tonn0297, 2019-05-05:
Recv: ok
Send: N12976 G1 X47.467 Y58.623 F9000.000*95
Recv: ok
Send: N12977 G1 F1800*77
Recv: ok
Send: N12978 G1 X47.003 Y58.430 E5.25396*99
Recv: ok
Send: N12979 G1 E3.25396 F2400.00000*17
Recv: ok
Send: N12980 G92 E0*69
Recv: ok
Send: N12981 G1 Z2.400 F9000.000*8
Recv: ok
Send: N12982 G1 X63.166 Y104.383 F9000.000*97
Recv: // Klipper state: Shutdown
Recv: ok
Send: N12983 G1 Z2.200 F9000.000*12
Recv: // MCU 'mcu' shutdown: No next step
Recv: // This is generally indicative of an intermittent
Recv: // communication failure between micro-controller and host.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: No next step
Changing monitoring state from "Printing" to "Cancelling"
Recv: ok
Send: N12984 G1 E2.00000 F2400.00000*25
Send: N12985 M108*45
Recv: // MCU 'mcu' shutdown: No next step
Recv: // This is generally indicative of an intermittent
Recv: // communication failure between micro-controller and host.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: No next step
Send: N12986 M84*27
Recv: ok
Send: N12987 M104 T0 S0*36
Recv: // MCU 'mcu' shutdown: No next step
Recv: // This is generally indicative of an intermittent
Recv: // communication failure between micro-controller and host.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: No next step
Recv: ok
Send: N12988 M108*32
Send: N12989 M140 S0*110
Recv: // MCU 'mcu' shutdown: No next step
Recv: // This is generally indicative of an intermittent
Recv: // communication failure between micro-controller and host.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: No next step
Recv: ok
Send: N12990 M106 S0*100
Recv: // MCU 'mcu' shutdown: No next step
Send: N12991 M108*40
Recv: // This is generally indicative of an intermittent
Recv: // communication failure between micro-controller and host.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: No next step
[... 216 lines cut ...]
Recv: !! MCU 'mcu' shutdown: No next step
Recv: ok
Send: M105
Recv: ok B:40.2 /40.0 T0:205.0 /205.0
Send: M105
Recv: ok B:40.1 /40.0 T0:204.2 /205.0
Send: M105
Recv: ok B:40.0 /40.0 T0:202.2 /205.0
Send: M105
Recv: ok B:39.9 /40.0 T0:198.8 /205.0
Send: M105
Recv: ok B:39.8 /40.0 T0:194.6 /205.0
Send: M105
Recv: ok B:39.7 /40.0 T0:190.4 /205.0
Send: M105
Recv: ok B:39.5 /40.0 T0:186.0 /205.0
Send: M105
Recv: ok B:39.4 /40.0 T0:181.9 /205.0
Send: M105
Recv: ok B:39.3 /40.0 T0:177.4 /205.0
Send: M105
Recv: ok B:39.3 /40.0 T0:173.3 /205.0
Send: M105
Recv: ok B:39.2 /40.0 T0:169.3 /205.0
Send: M105

KevinOConnor, 2019-05-08:
You've set your baud rate to 19200.  That's amusing, but it's certainly not recommended.  Return Klipper to its default baud of 250000 for better results.

-Kevin

tonn0297, 2019-05-08:
I have a mks 1.4 that don't support more..

jakep82, 2019-05-08:
Yes it does.  I've used an MKS Gen 1.4 at the recommended 250000 baud with zero issues.

closed 2019-05-25 by KevinOConnor