Klipper3d/klipper #1610: Rotary delta ( Delta robot kinematics )

mapa3m1, 2019-05-05:
Dear Sir,
Good day,

I am sure that you know about such kinematics as a rotating delta. But at the moment as far as I know the klipper does not support this kinematics.
Could you please add this kinematics to the klipper?
I found on the Internet some resources with kinematics calculations and examples of source code in С language.
But my knowledge of programming languages ​​and mathematics is not enough to do this properly.
I am sure that many people will use this kinematics and will support you.

But there is a nuance, an article with calculations of kinematics in Russian. So help us Google translator.

[https://habr.com/en/post/390281/](url)
[https://www.marginallyclever.com/other/samples/fk-ik-test.html](url)

> // dimensions of the robot
> // (designations see on the scheme)
> const float e = 115.0;     // side of the movable platform
> const float f = 457.3;     // fixed base side
> const float re = 232.0;
> const float rf = 112.0;
> 
> // trigonometric constants
> const float sqrt3 = sqrt(3.0);
> const float pi = 3.141592653;    // PI
> const float sin120 = sqrt3/2.0;
> const float cos120 = -0.5;
> const float tan60 = sqrt3;
> const float sin30 = 0.5;
> const float tan30 = 1/sqrt3;
> 
> // direct kinematics: (theta1, theta2, theta3) -> (x0, y0, z0)
> // return status: 0=OK, -1=nonexistent position
> int delta_calcForward(float theta1, float theta2, float theta3, float &x0, float &y0, float &z0) {
>     float t = (f-e)*tan30/2;
>     float dtr = pi/(float)180.0;
> 
>     theta1 *= dtr;
>     theta2 *= dtr;
>     theta3 *= dtr;
> 
>     float y1 = -(t + rf*cos(theta1));
>     float z1 = -rf*sin(theta1);
> 
>     float y2 = (t + rf*cos(theta2))*sin30;
>     float x2 = y2*tan60;
>     float z2 = -rf*sin(theta2);
> 
>     float y3 = (t + rf*cos(theta3))*sin30;
>     float x3 = -y3*tan60;
>     float z3 = -rf*sin(theta3);
> 
>     float dnm = (y2-y1)*x3-(y3-y1)*x2;
> 
>     float w1 = y1*y1 + z1*z1;
>     float w2 = x2*x2 + y2*y2 + z2*z2;
>     float w3 = x3*x3 + y3*y3 + z3*z3;
> 
>     // x = (a1*z + b1)/dnm
>     float a1 = (z2-z1)*(y3-y1)-(z3-z1)*(y2-y1);
>     float b1 = -((w2-w1)*(y3-y1)-(w3-w1)*(y2-y1))/2.0;
> 
>     // y = (a2*z + b2)/dnm;
>     float a2 = -(z2-z1)*x3+(z3-z1)*x2;
>     float b2 = ((w2-w1)*x3 - (w3-w1)*x2)/2.0;
> 
>     // a*z^2 + b*z + c = 0
>     float a = a1*a1 + a2*a2 + dnm*dnm;
>     float b = 2*(a1*b1 + a2*(b2-y1*dnm) - z1*dnm*dnm);
>     float c = (b2-y1*dnm)*(b2-y1*dnm) + b1*b1 + dnm*dnm*(z1*z1 - re*re);
> 
>     // discriminant
>     float d = b*b - (float)4.0*a*c;
>     if (d < 0) return -1; // nonexistent position
> 
>     z0 = -(float)0.5*(b+sqrt(d))/a;
>     x0 = (a1*z0 + b1)/dnm;
>     y0 = (a2*z0 + b2)/dnm;
>     return 0;
> }
> 
> // inverse kinematics
> // auxiliary function, angle calculation theta1 (in the  YZ plane )
> int delta_calcAngleYZ(float x0, float y0, float z0, float &theta) {
>     float y1 = -0.5 * 0.57735 * f; // f/2 * tg 30
>     y0 -= 0.5 * 0.57735 * e;       // move the center to the edge
>     // z = a + b*y
>     float a = (x0*x0 + y0*y0 + z0*z0 +rf*rf - re*re - y1*y1)/(2*z0);
>     float b = (y1-y0)/z0;
>     // discriminant
>     float d = -(a+b*y1)*(a+b*y1)+rf*(b*b*rf+rf);
>     if (d < 0) return -1; // nonexistent point
>     float yj = (y1 - a*b - sqrt(d))/(b*b + 1); // choose an external point
>     float zj = a + b*yj;
>     theta = 180.0*atan(-zj/(y1 - yj))/pi + ((yj>y1)?180.0:0.0);
>     return 0;
> }
> 
> // inverse kinematics: (x0, y0, z0) -> (theta1, theta2, theta3)
> // return status: 0=OK, -1=nonexistent position
> int delta_calcInverse(float x0, float y0, float z0, float &theta1, float &theta2, float &theta3) {
>     theta1 = theta2 = theta3 = 0;
>     int status = delta_calcAngleYZ(x0, y0, z0, theta1);
>     if (status == 0) status = delta_calcAngleYZ(x0*cos120 + y0*sin120, y0*cos120-x0*sin120, z0, theta2);  // rotate coords to +120 deg
>     if (status == 0) status = delta_calcAngleYZ(x0*cos120 - y0*sin120, y0*cos120+x0*sin120, z0, theta3);  // rotate coords to -120 deg
>     return status;
> }

Thank you.

Hywelmartin, 2019-05-14:
if anyone give it a go.. a moving bed would be awesome...

KevinOConnor, 2019-05-16:
FWIW, this looks like the same type of kinematics that is implemented by smoothieware.  The Kinematics themselves are not hard - the difficulty is in testing it.  Do you have the particular printer?  Have you been able to install Klipper on it and test that basic motor functions work?

-Kevin

mapa3m1, 2019-05-17:
> FWIW, this looks like the same type of kinematics that is implemented by smoothieware. The Kinematics themselves are not hard - the difficulty is in testing it. Do you have the particular printer? Have you been able to install Klipper on it and test that basic motor functions work?
> 
> -Kevin

I would like to build it, spare part already shipped, i am already use klipper on my cartesian printer.
About testing that basic motor functions work - yes, but a little bit later, now i'm so far from my home.

I know that many people from the Russian-speaking community want to have a printer with such kinematics, but the problem is that there is no software for this kinematics.
There is one person who made a printer with a kinematics delta robot and uses smoothieware.  [https://www.thingiverse.com/AlexKorvin/about](url) 
But smoothieware has some bugs.

KevinOConnor, 2019-05-25:
I'm going to mark this as inactive then as it doesn't look like anyone is currently working on an implementation for it.

-Kevin

A1exKorvin, 2019-10-13:
Hi my name is Alex Korvin
I made a Non-linear Delta 3D Printer
https://www.thingiverse.com/thing:3489886 
I want to make it work with klipper
could you please port the kynematics for it from the smoothieware firmware?
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/robot/arm_solutions/RotaryDeltaSolution.cpp 
And here you can find the Marlin firmware, adapted for the delta robot. Perhaps this will help. 
https://github.com/TTN-/IcePick-Delta/tree/master/Firmware
Here you can read a little about the kinematics of the delta robot. 
http://forums.trossenrobotics.com/tutorials/introduction-129/delta-robot-kinematics-3276/

KevinOConnor, 2019-10-15:
I can help with the basic implementation of the rotary delta kinematics.  However, it would require someone with the hardware to test and provide feedback.

If you're willing to do that, please start by installing klipper on the machine - use simple cartesian kinematics to start and verify that all the main functionality is working.  Use the STEPPER_BUZZ command (as described in the installation and config checks documents) to verify that all the motors are working properly.  Then attach the resulting klipper log file here.  I will also need to know the key parameters of the machine (arm lengths, platform height, platform radius, effector radius, etc.).

-Kevin

A1exKorvin, 2019-10-18:
> I can help with the basic implementation of the rotary delta kinematics. However, it would require someone with the hardware to test and provide feedback.
>
> If you're willing to do that, please start [...]
Ok, thanks, I’ll definitely do it =)

A1exKorvin, 2019-10-18:
> Alex thank you for reaching out if you need help. Do not hesitate to contact me. Or come over the klipper discord channel. they are some russian speakers here and there.

Thanks =)

KevinOConnor, 2019-11-13:
Any further progress with this?

-Kevin

maketo3D, 2019-11-17:
It would be very interesting, I was thinking of making one but there are hardly any firmwares that control it.. I wish I could work with her together with klipper.

KevinOConnor, 2019-11-20:
I put together code for the rotary delta kinematics:

cd ~/klipper ; git fetch ; git checkout origin/work-rotary-20191119 ; sudo service klipper restart

This is preliminary code (hasn't been tested on real hardware; no boundary checks are implemented).  If anyone tests this, please report your findings (success or failure).

-Kevin

A1exKorvin, 2019-11-20:
> I put together code for the rotary delta kinematics:
>
> cd ~/klipper ; git fetch ; git checkout origin/work-rotary-20191119 ; sudo service klipper restart
>
> This is preliminary code (hasn't been tested [...]

Hi. Sorry for the long reply. I am currently busy with ongoing projects and there is no time to conduct tests yet. I will ask for help with this guys who already have a assembled robot =)

vladT0, 2019-11-25:
Hi Kevin !  
Please see requested files. I have a chance only to test XYZ motors and endstops.
They are working correct. I can not say it was exactly 1mm, but seems so.
Hope it will help. I am ready to send you any additional information if I missed something.
Regards,
Vlad.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3888963/klippy.log)

[printer.txt](https://github.com/KevinOConnor/klipper/files/3888968/printer.txt)

KevinOConnor, 2019-11-25:
Great.  The next step is to pull down the developer branch with the rotary delta code:

cd ~/klipper ; git fetch ; git checkout origin/work-rotary-20191119 ; sudo service klipper restart

Then you'll need to update your config based on the example given in config/example-rotary-delta.cfg .  Let me know if you have questions about the config.

-Kevin

Hywelmartin, 2019-11-27:
Interesting...
Would it be difficult to add a z-axis to overcome that limitation.. I think it would be an interesting idea for normal deltas as well..

vladT0, 2019-11-27:
@KevinOConnor 

Hi Kevin !
Thanks a lot for your very fast response.  
I followed your instructions, but seems made something wrong.
1) Please see attached pic with typical delta arms IDs. I am a little bit confused. According your
comments in example config file, you have other IDs for arms. I show it in bold find with word 
Klipper.  Usually IDs named counterclock-wise.

2) I am not sure, I calculated correctly step_distance.
For Smoothie it was:
LV8729  driver set for 128 microsteps = (200 * 128 * 26.01)/360 = 1805
Klipper: I take formula from your comments and just added reduction 26.01
2*3.1415*26.01*1.8/360/128 = 0.0063836262
First time I calculate it as
1/1805 = 5.54016620498615e-4

3) G28 command is not working. See logs.

Thanks in advance for your help.

Vlad.

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/3899045/klippy.log)

[printer.txt](https://github.com/KevinOConnor/klipper/files/3899048/printer.txt)

KevinOConnor, 2019-11-28:
Yes, I think your arm diagram is correct.  We can change Klipper to match Smoothie, but lets leave it like this until we get it working.

If there is a 26.01 gearing ratio on a 1.8 degree stepper with a 128 microsteps, then you would use `2 * pi * (1.8 / 360) / 128 / 26.01 == 0.000009436252444`.  (This is also quirky and should be changed in Klipper, but lets see if it works first.)

> (200 x 128 x 26.01)/360 = 1805

FYI, (200 * 128 * 26.01)/360 = 1849.6 - I'm not sure if that was a typo or if you have a different gearing ratio.

-Kevin

vladT0, 2019-11-28:
Hi Kevin ! 
It was mistake, I have to divide by gearing ratio :)
Correct figures for me are:
2 * pi * (1.8 / 360) / 128 / 25.3835 = 0.00000966886708
(200 x 128 x 25.3835)/360 = 1805
I will check today, I see only one problem, according comments in config, I have to use just 6 digits,
in this case I can take 0.000009 and you will lose precision in calculation.

Regards,
Vlad.

KevinOConnor, 2019-11-28:
> according comments in config, I have to use just 6 digits,

For a rotary delta, I'd use 15 digits.

-Kevin

KevinOConnor, 2019-11-28:
Just for my knowledge, what gearing are you using for the steppers?  Is it a two gear system (small gear on the stepper and large gear on the shoulder joint)?  If so, what's the tooth count and belt size (eg GT2) on each of the pulleys?

-Kevin

vladT0, 2019-11-28:
I attached here screenshot from Alex's video.
[image]
About gearing ratio: can I use 15 digits right now or it will be in the next versions ?

KevinOConnor, 2019-11-28:
> can I use 15 digits right now or it will be in the next versions ?

Use 15 digits with the current code.

-Kevin

vladT0, 2019-11-28:
Hi Kevin !
Please see attached files. With this config motors are not working. After command STEPPER_BUZZ STEPPER=stepper_a firmware crashed. I think you see this event in the log.  I can reload it only with Reset button on printer controller.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3902953/klippy.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/3902954/printer.txt)

Regards,
Vlad.

KevinOConnor, 2019-11-29:
There was an error in the Klipper code related to step_distance being in radians (vs the normal millimeters) which caused both the endstop checking and stepper_buzz code to produce incorrect timing.  I've put together a workaround on the work-rotary-20191119 branch:

cd ~/klipper ; git fetch ; git checkout origin/work-rotary-20191119 ; sudo service klipper restart

-Kevin

KevinOConnor, 2019-11-29:
Thanks for the diagram.  I'm still a little confused on the gearing ratio.  The diagram seems to show a 60:16 gear driving a 107:16 gear, which results in a 25.078125 gearing ratio (`60*107 / (16*16) = 25.078125`).  (I'm guessing the 25.068 is different just due to rounding differences.)

You're reporting a 25.3835 gearing ratio, however.  Do you have a 60:16 gear followed by a 108:16 gear (for a gearing ratio of `60*108 / (16*16) = 25.3125`)?

Or, does the final pulley not have teeth (or not have a whole number of teeth)?

Also, what type of host machine are you running the Klipper software on?

Cheers,
-Kevin

vladT0, 2019-11-29:
Hi Kevin ! 
I will install right now new version and test it.
About gearing ratio: 
I think from this screenshot, from Alex's video, will be clear. Yes, you are right, final pulley
1) Printed 
2) Without teeth.
I have a little bit other size of pulley.
Host is Raspbery Pi (the first version ).

[image]

vladT0, 2019-11-29:
#The steps per degree are calculated as:
#  For Firepick delta robot: 
# (1) First determine the circumference of the big pulley, which has a smooth surface, and calculate the 'teeth it would have'
# (2) Determine the GT2 belt thickness and tooth depth
# (3) Calculate the ACTUAL big pulley circumference taking into account that the GT2 belt affects this as
#     it rests on the big pulley and not into teeth cut into the pulley, the formula is:
#     BIG_PULLEY_CIRCUMFERENCE = BIG_PULLEY_TEETH * TOOTH_SPACING + (BELT_THICKNESS - TOOTH_DEPTH)*6.283185
# (4) Calculate the ACTUAL small pulley circumference taking into account that the GT2 belt affects this as
#     rests into the small pulley as it has teeth cut into the pulley, the formula is:
#     SMALL_PULLEY_CIRCUMFERENCE = SMALL_PULLEY_TEETH * TOOTH_SPACING
# (5) Now determine the pulley reduction which is:
#     PULLEY_REDUCTION = BIG_PULLEY_CIRCUMFERENCE / SMALL_PULLEY_CIRCUMFERENCE
# (6) Finally, given the number of steps a stepper motor needs for a full 360 degree rotation, and the number of microsteps your
#     stepper motor driver chip is set for, you can calculate the STEPS_PER_MM for each axis, which for a Rotary Delta, are the same:
#    [ALPHA/BETA/GAMMA]_STEPS_PER_MM = (XYZ_STEPS_PER_ROTATION*XYZ_MICROSTEPS*PULLEY_REDUCTION)/360

vladT0, 2019-11-29:
Hi Kevin ! 
Nothing fetched for me. All files are the same. 
work-rotary-20191119 - is it correct branch ?
Regards,
Vlad.

KevinOConnor, 2019-11-29:
Oops.  Forgot to push.  Should be up now.  Also, note that STEPPER_BUZZ should now move the shoulder joint back and forth 1 radian instead of 1mm.

-Kevin

vladT0, 2019-11-29:
Hi Kevin ! 
G28 was working to the wrong dimension. I inverted DIR pins and seems G28 is working correctly now.  Homing_retract is not tune-able and executed with the same speed. I would start retract much slower. It helps to find more precision home position.
So I executed G28, check positions with M114 and tried G0 Z200 and
mcu crashed again. See log. Please advice, what to execute for the next tests.
Regards,
Vlad.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3906495/klippy.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/3906496/printer.txt)

KevinOConnor, 2019-11-29:
The second home should occur at half the homing speed - it can be tuned via the `second_homing_speed` parameter (see config/example.cfg).

My first impression of the error is that the rpi1 got overloaded (looks like it took 400ms to produce the first 500ms worth of steps and that caused the micro-controller to trigger an error).  Klipper is better on an rpi2 (or faster) host machine.  That said, generating a z-only move while the head is near the top of the frame requires the most work.  You could try first moving the head down with a slow move (eg, `G0 Z100 F300`), to see if moving out of that trouble zone improves your results.

Otherwise, I'll try to take a closer look at the error over the next couple of days.

-Kevin

vladT0, 2019-11-30:
Hi Kevin !
You was completely right. It was issue with host. F300 options solved issue. I will change the host next week to the new one, version 3 or 4. Is Zero faster ? I have it at home too.
Very good news, distance of the Z moving is very closer to the real mm size, I measured it. I never could get this result with Smoothie, don't know why, but it is other story.
Could I test something also with Pi1 host ? or shall I wait before you improve source code for the delta ? 
Minor bug, second homing speed was not recognized. See log file. 
Regards,
Vlad.

KevinOConnor, 2019-11-30:
> Is Zero faster ?

I don't think it is faster.  (Or, only slightly faster.)

> Could I test something also with Pi1 host ?

Do X and Y moves work properly?  As long as you avoid fast Z only moves, I suspect the pi1 will work okay.  You could also go to 16 microsteps on the stepper drivers to reduce the load.  Finally, you could try changing the `MOVE_BATCH_TIME = 0.500` in klippy/toolhead.py to `MOVE_BATCH_TIME = 0.200`.  (If changing the host source code, be sure to do a `sudo service klipper restart` after every change.)

> Minor bug, second homing speed was not recognized.

Looks like a colon was missing in the config file - it should be: `second_homing_speed: 10`

-Kevin

vladT0, 2019-11-30:
1) About MOVE_BATCH_TIME - what is sense of this param ? Is it time for affected on communication between host and microcode kernel ? 
2) I have tmc2208, they are working in simple mode as 1/16, but I noticed Klipper supported special modes for it, can I test it with Rotary delta or it is not implemented right now ?

KevinOConnor, 2019-11-30:
MOVE_BATCH_TIME is an internal accounting setting.  It controls how many steps the host may generate before flushing them to the micro-controller.  It doesn't alter the robot behaviour.

The tmc2208 needs to be have the UART line wired up and `[tcm2208 stepper_a]` type sections added to the config.  Otherwise, it should work fine.

-Kevin

vladT0, 2019-11-30:
Good evening Kevin  ! 
I changed MOVE_BATCH_TIME = 0.200 and now G0 command is working with usual speed and did not crash.  X and Y are working correct too. Unfortunately second_homing_speed: 10 has no affect on speed. I expect homing_retract_dist: 10.0 moves back very slowly.  10mm or 10 radian  ?
Regards,
Vlad.

vladT0, 2019-12-01:
> The tmc2208 needs to be have the UART line wired up and `[tcm2208 stepper_a]` type sections added to the config. Otherwise, it should work fine

Hi Kevin !
I have tried to use TMC2208 V2.0, what I did step by step:
1) Recalculate step per radiant according 1/16.
2) Solder pad for all drivers.
3) Removed all jumper caps under the driver.
4) Put jumper caps to the XUART, YUART, ZUART.
5) Configure UART section for each stepper. Here I was a little bit confused, what pin to use, RX or TX, but seems according example with SKR board was used RX pin.
6) According board documentation, if pad is soldered, I don't need any additional wire.
If using the TMC2208 V2.0 version, you don’t need to connect NC and uart on the driver.

Service was restarted, but klipper can not communicate with UART pin, See log. Please advice, the reason, what is mistake ?
[klippy.log](https://github.com/KevinOConnor/klipper/files/3908471/klippy.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/3908472/printer.txt)

KevinOConnor, 2019-12-01:
> I changed MOVE_BATCH_TIME = 0.200 and now G0 command is working with usual speed and did not crash. X and Y are working correct too.

Great.

> Unfortunately second_homing_speed: 10 has no affect on speed. I expect homing_retract_dist: 10.0 moves back very slowly. 

I don't see anything wrong with the code.  Try setting `homing_retract_dist: 10` and `second_homing_speed: .5` to emphasize the effect.  Note that the second homing speed only impacts the second move towards the endstop switch, it does not reduce the speed of the move away from the endstop switch.

> 10mm or 10 radian ?

Homing speeds are in mm/s.

> Service was restarted, but klipper can not communicate with UART pin, See log.

Make sure motor power is enabled when starting Klipper.  Also, the pins in your config don't seem to match the pins in the default `generic-bigtreetech-skr-v1.3.cfg` so you should double check them.

-Kevin

vladT0, 2019-12-01:
> Make sure motor power is enabled when starting Klipper. Also, the pins in your config don't seem to match the pins in the default `generic-bigtreetech-skr-v1.3.cfg` so you should double check them.
> 
> -Kevin

I have SGEN_L board, very similar to bigtreetech-skr-v1.3, but pins are other.  I configured **RX** pins. See picture.
[image]

KevinOConnor, 2019-12-01:
The most common reason for the TMC2208 to not identify is the board not having motor power during startup.  The other common failure is when the UART pin is not correctly routed to the micro-controller.  Your pin assignment looks okay according to the given picture.  However, some tmc2208 stepsticks may use a different pin for pdn-uart than the board is expecting, and/or the stepstick may require a solder jumper to be soldered.

You may wish to comment out all but one of the tmc2208 config sections and see if you can get any of the drivers to configure.

The work-rotary-20191119 branch has all recent changes.  The tmc2208 code has no relation to the rotary delta kinematics so it seems unlikely they are related.

-Kevin

KevinOConnor, 2019-12-02:
FYI, I've updated the work-rotary-20191119 branch.  The latest code changes the default stepper angles (stepper_a is now at 30 degrees, b at 150, and c at 270).  I've also added some basic boundary checking to the code.

cd ~/klipper ; git fetch ; git checkout origin/work-rotary-20191119 ; sudo service klipper restart

-Kevin

vladT0, 2019-12-02:
@KevinOConnor : you was right. UART Init issue was fixed, I added comment here https://github.com/KevinOConnor/klipper/issues/542

I tested new version.  Stepper_a is now at 30 degrees, b at 150, and c at 270 : confirmed.
New issue. Real distance Z ( as example ) is far enough from G0 Z0; G0 Z100; I don't know the reason : tmc2208 in UART mode or anything else. Seems step_distance is correct. Do you have idea ?
I understand how 1/16 static step is working, but I am not sure I understand  UART mode.

KevinOConnor, 2019-12-03:
> New issue. Real distance Z ( as example ) is far enough from G0 Z0; G0 Z100; I don't know the reason

Sorry, I didn't understand that.

> I tried M122 command, but I assume it is not implemented in Klipper.

You can use the DUMP_TMC command (see docs/G-Codes.md for details).  You're config looks okay to me, though.  You'll likely need to get the tmc2208 specifications from Trinamic in order to read the DUMP_TMC output.

-Kevin

vladT0, 2019-12-03:
Sorry for the bad explanation. 
1) G28; G0 Z0;  and measure distance from the nozzle to the table. 
2) G0 Z100;  Expect nozzle distance is 100mm from the point 1, but it is a bit less.
With LV8729 and static 1/128 micro-steps result was very closer to 100mm.

KevinOConnor, 2019-12-03:
> Expect nozzle distance is 100mm from the point 1, but it is a bit less.

That sounds like a calibration issue.  I don't know why it would be different with the lv8729.

If the upper_arm_length is not 170mm, if the lower_arm_length is not 320mm, or if the distance from the effector joint to the plane holding the shoulder joints is not 160.9mm (412.9 - 252) when the endstop triggers then it will result in Z moves being skewed.

If you run GET_POSITION after each movement, what does it report and what are the real distances?

> I am going to simulate printing. I mean print something without filament.
Will it work ?

As far as I know it should work.

-Kevin

vladT0, 2019-12-03:
Hi Kevin !
GET_POSITION was very helpful for me.
1) G28 
```
Send: GET_POSITION
Recv: // mcu: stepper_a:-9 stepper_b:-6 stepper_c:3
Recv: // stepper: stepper_a:0.687797 stepper_b:0.687797 stepper_c:0.687797
Recv: // kinematic: X:0.000000 Y:-0.000000 Z:252.000000
Recv: // toolhead: X:0.000000 Y:-0.000000 Z:252.000000 E:0.000000
Recv: // gcode: X:0.000000 Y:-0.000000 Z:252.000000 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
```
2) G0 Z0

```
Send: GET_POSITION
Recv: // mcu: stepper_a:-20014 stepper_b:-20011 stepper_c:-20002
Recv: // stepper: stepper_a:-0.859654 stepper_b:-0.859654 stepper_c:-0.859654
Recv: // kinematic: X:0.000000 Y:-0.000000 Z:-0.006265
Recv: // toolhead: X:0.000000 Y:-0.000000 Z:0.000000 E:0.000000
Recv: // gcode: X:0.000000 Y:-0.000000 Z:0.000000 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
```
Am I right, if in ideal case after G28 command line mcu: has equal figures for each stepper ?
Could you please explain, what does it mean here positive, negative and null position here ?
I can imagine, if null position is when any of the endstops is triggered and you calculate position of other two how shift from it. But I am confused in this case with negative values.

KevinOConnor, 2019-12-04:
I've added some information on GET_POSITION to the documentation: https://www.klipper3d.org/Code_Overview.html#coordinate-systems

> Am I right, if in ideal case after G28 command line mcu: has equal figures for each stepper ?

No, it is normal for the "mcu" positions to differ.  For example, if one stepper just happens to move a little further than another stepper during a G28 then the mcu positions would differ.

> Could you please explain, what does it mean here positive, negative and null position ?

Hopefully the link above helps with that.

> I can imagine, if null position is when any of the endstops is triggered and you calculate position of other two how shift from it. But I am confused in this case with negative values.

For the mcu position, negative just means the micro-controller has stepped more in a clockwise direction than a counterclockwise direction since it was last reset.  It doesn't really mean much.

For the stepper position, it's the current angle (in radians) of the stepper (as tracked by the kinematics code).  It's negative here, because if you looked at one of the stepper motor joints the angle would indeed be negative.

The kinematic Z position is negative because the request to move to Z=0 could only be approximated due to the discrete nature of the stepper motors.  The closest Klipper could get to Z=0 was actually 6 microns lower (at least as far as the kinematics code is tracking the position).

What were the actual measured positions during these attempts?

-Kevin

vladT0, 2019-12-04:
1) G28
Real nozzle position Z = 230
```
Recv: // mcu: stepper_a:-10 stepper_b:4 stepper_c:6
Recv: // stepper: stepper_a:0.687797 stepper_b:0.687797 stepper_c:0.687797
Recv: // kinematic: X:0.000000 Y:-0.000000 Z:252.000000
Recv: // toolhead: X:0.000000 Y:-0.000000 Z:252.000000 E:0.000000
Recv: // gcode: X:0.000000 Y:-0.000000 Z:252.000000 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
``` 
2)  G0 Z0
Real nozzle position Z = 0
```
Recv: // mcu: stepper_a:-20015 stepper_b:-20001 stepper_c:-19999
Recv: // stepper: stepper_a:-0.859654 stepper_b:-0.859654 stepper_c:-0.859654
Recv: // kinematic: X:0.000000 Y:-0.000000 Z:-0.006265
Recv: // toolhead: X:0.000000 Y:-0.000000 Z:0.000000 E:0.000000
Recv: // gcode: X:0.000000 Y:-0.000000 Z:0.000000 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok
```
3)  G0 Z100
Real nozzle position Z=102mm

```
Recv: // mcu: stepper_a:-13625 stepper_b:-13611 stepper_c:-13609
Recv: // stepper: stepper_a:-0.365367 stepper_b:-0.365367 stepper_c:-0.365367
Recv: // kinematic: X:0.000000 Y:-0.000000 Z:100.007842
Recv: // toolhead: X:0.000000 Y:-0.000000 Z:100.000000 E:0.000000
Recv: // gcode: X:0.000000 Y:-0.000000 Z:100.000000 E:0.000000
Recv: // gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000
Recv: // gcode homing: X:0.000000 Y:0.000000 Z:0.000000
Recv: ok

```

vladT0, 2019-12-04:
Hi Kevin !
I have made measures for 3 positions in the prev. message. I am not sure, if nozzle physically is in the center of the heating bed. You know, how to calibrate delta. The step one, you have to calibrate endstops, when nozzle goes down exactly to the (0,0,0) position. I had impression, I can use mcu: info from GET_POSITION command for endstop calibration.
Regards,
Vlad.

KevinOConnor, 2019-12-05:
Was this printer previously working on Smoothieware?  If so, can you post the working Smoothie config?

If this is the first time getting the printer working then calibration will be needed.  Unfortunately, I'm not familiar with the steps needed to calibrate a rotary printer.  It would be nice to get the DELTA_CALIBRATE command working on rotary delta printers (as the calibration it does should be similar).

At a minimum, if the real distance from the tip of the nozzle to bed is 230mm after a G28, then position_endstop should be set to 230mm.  (It may also then be necessary to then tune shoulder_height.)

Separately, I've updated the work-rotary-20191119 branch to improve the STEPPER_BUZZ handling.  Can you check if everything works okay on it?

cd ~/klipper ; git fetch ; git checkout origin/work-rotary-20191119 ; sudo service klipper restart

-Kevin

vladT0, 2019-12-05:
> Was this printer previously working on Smoothieware? If so, can you post the working Smoothie config?
Unfortunately not. I have started only tests with  Smoothieware too, but for me in  Smoothie was a lot of unclear things without any support from  Smoothie team side.

> At a minimum, if the real distance from the tip of the nozzle to bed is 230mm after a G28, then position_endstop should be set to 230mm. (It may also then be necessary to then tune shoulder_height.)
I see, this 252 came from Smothieware, where this distance had a little bit other sense, it is max travel for Z.  I will create today photo of the printer and will show all measures, may be you can give 
me advice where I am wrong.  New version I will check too.

KevinOConnor, 2019-12-05:
I've added initial support for the DELTA_CALIBRATE tool to the rotary delta kinematics.

To use this tool:
* Grab the latest code: `cd ~/klipper ; git fetch ; git checkout origin/work-rotary-20191119 ; sudo service klipper restart`
* Add a `[delta_calibrate]` config section to the printer.cfg file.  See the updated config/example-rotary-delta.cfg file for an example.
* Set `horizontal_move_z=40` in `[delta_calibrate]` and `minimum_z_position=-40` in `[printer]` (or similarly large values).
* Run `G28` to home the printer.
* Measure the distance from the tip of the nozzle to the bed.  Then enter the following command (make sure to not move the head between measuring and entering this command): `DELTA_ANALYZE MANUAL_HEIGHT=<measured height>`
* Run `DELTA_CALIBRATE METHOD=manual`.  Do not issue a restart or otherwise change the config between this step and the previous step.  Follow the directions at https://www.klipper3d.org/Delta_Calibrate.html to perform the basic delta calibration.
* At the completion of the basic calibration run the `SAVE_CONFIG` command.
* Attach the resulting klipper.log file to this issue (success or failure).

Ideally the above will provide accurate values for your position_endstop and shoulder_height settings.  After running the steps above, you should return the `horizontal_move_z` and `minimum_z_position` settings back to their defaults.

-Kevin

vladT0, 2019-12-06:
> Separately, I've updated the work-rotary-20191119 branch to improve the STEPPER_BUZZ handling. Can you check if everything works okay on it?

It is working correctly, but distance of the moving in the old version was around (+/-)1mm, in the new version it is much bigger.

vladT0, 2019-12-06:
> I've added initial support for the DELTA_CALIBRATE tool to the rotary delta kinematics.

Seems new version has a bug in source code, G28 is not working anymore.
[klippy.1.log](https://github.com/KevinOConnor/klipper/files/3933383/klippy.1.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/3933384/printer.txt)

KevinOConnor, 2019-12-07:
It doesn't like the `shoulder_height = 390.8` that you recently changed.  I'll have to look at why, but for now, I'd go back to 420 and see what delta_calibrate recommends.

-Kevin

vladT0, 2019-12-07:
I have frame from 20x20 alu profile and actually moved down a little bit upper plate. So 390.8 is correct size now, but we can try again with 420.

vladT0, 2019-12-07:
Hi Kevin !
I finished calibration procedure according your document. Please see attached files.  Optimized parameters are very close to measured. I understand how calibration is working now and I am going to repeat it again more precise. Thanks a lot for your help with rotary delta. I like Klipper !!! 
Regards,
Vlad.

KevinOConnor, 2019-12-08:
> shoulder_height = delta_z_offset - delta_ee_offs - delta_tool_offset

Yes.  From the diagram, `position_endstop` is `<distance8>` and `shoulder_height` is `<distance5> - (<distance6> + <distance7>)`.

> I finished calibration procedure according your document.

The log indicates that the `DELTA_ANALYZE MANUAL_HEIGHT=<measured height>` step did not work properly.  I suspect you will need to perform that step to get a good calibration on a rotary delta printer.  Try running the steps again at https://github.com/KevinOConnor/klipper/issues/1610#issuecomment-562364335 - that is, run G28, measure the distance from nozzle tip to bed, run `DELTA_ANALYZE MANUAL_HEIGHT=<measured distance>`, run `DELTA_CALIBRATE METHOD=manual`, run `SAVE_CONFIG` and then attach the log here.

If the printer is working and you are able to print basic objects, you could try running the "extended delta calibration" described at: https://www.klipper3d.org/Delta_Calibrate.html

-Kevin

vladT0, 2019-12-08:
Hi Kevin !
I will repeat it again. The measured distance from the bed after G28 was the same and I had
wrong decision , I don't need in this case 
DELTA_ANALYZE MANUAL_HEIGHT=<measured distance>
command.

Other issue. 
G28 command is not working in all cases.
Example.

```
Send: M114
Recv: X:0.000 Y:70.000 Z:0.000 E:0.000
Recv: ok
Send: G28
Recv: !! Failed to home c: Timeout during endstop homing
Changing monitoring state from 'Operational' to 'Error:  Failed to home c: Timeout during endstop homing\x0a'
Recv: ok
```

It is very easy reproduce. If any coordinates is not very near to other two. One endstop triggered, second too, but last one need to travel. Instead of this we have this message.

vladT0, 2019-12-08:
> The log indicates that the `DELTA_ANALYZE MANUAL_HEIGHT=<measured height>` step did not work properly. I suspect you will need to perform that step to get a good calibration on a rotary delta printer. Try running the steps again at [#1610 (comment)](https://github.com/KevinOConnor/klipper/issues/1610#issuecomment-562364335) - that is, run G28, measure the distance from nozzle tip to bed, run `DELTA_ANALYZE MANUAL_HEIGHT=<measured distance>`, run `DELTA_CALIBRATE METHOD=manual`, run `SAVE_CONFIG` and then attach the log here.

I have done it exactly according document. Please see attached files.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3937182/klippy.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/3937183/printer.txt)

[23 more replies]

closed 2020-01-23 by KevinOConnor