Klipper3d/klipper #1613: [Feature Request] Conditional macros

tameike000, 2019-05-06:
Hello.
I am looking for a way to bring some logic into my macro's:
Example:
`print_start filament=PLA`

```
[gcode_macro print_start]
if filament == PLA
    macro_PLA
if filament == ABS
    macro_ABS
else
    macro_youaredoomed
```
Maybe I am not the only one and this could be considered as a future feature.

Thanks.
M

Hywelmartin, 2019-05-07:
you know that you can have parameters in macros, that can be changed when you send the command.

If I want the toolhead to park in an other spot than the normal one I send 

PARK_MACRO Y=85 Z=50

[gcode_macro PARK_MACRO]
default_parameter_X: 0
default_parameter_Y: -85
default_parameter_Z: 30
gcode:
    M117 PARKING..
    G91
    G1 E-3.14 F1500
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F3000

tameike000, 2019-05-07:
@Hywelmartin 
Thanks so much for that example! I think I did not fully understand the potential of the existing implemention yet.

mcmatrix, 2019-05-07:
Actually, I'll tell you a little secret. 
You can use the Jinja2 templating engine in `gcode_macros` gcode.
The latest code in the master branch has it built in.
http://jinja.pocoo.org/docs/2.10/templates/

```ini
[gcode_macro print_start]
default_parameter_filament: PLA
gcode:
    {% if filament == 'PLA' %}
        macro_PLA
    {% elif filament == 'ABS' %}
        macro_ABS
    {% else %}
        macro_youaredoomed
    {% endif %}
```
This example is not tested in working klipper.

tameike000, 2019-05-07:
@mcmatrix if that works I'll send 1001 brownie points over.
Thanks for the hint. I will dig into that.

KevinOConnor, 2019-05-25:
I'm going to close this ticket as it looks like the conversation has concluded.  As indicated, conditional macros are supported (though documentation is still in progress).

-Kevin

closed 2019-05-25 by KevinOConnor

SrGeek, 2019-07-11:
@mcmatrix will the following work?
[gcode_macro park]
default_parameter_tool:0
gcode:
    {% if TOOL== 0 %}
        G1 X-12 F6000
    {% else %}
        G1 X475 F6000
    {% endif %}

or do I need elif in order to make it work?