Klipper3d/klipper #1614: Permission denied on Logfile Rollover

sniperlucian, 2019-05-06:
klippy lately changed log file location form /tmp/klippy.log  to  /var/log/klippy.log.

However Python fails to make an log rotate if the log files gets too big - guess this might happen also during a long print - making it fail. 

May 06 15:02:57 serv python[18604]: Logged from file tmc5160.py, line 463
May 06 15:02:57 serv python[18604]: Traceback (most recent call last):
May 06 15:02:57 serv python[18604]:   File "/usr/lib/python2.7/logging/handlers.py", line 77, in emit
May 06 15:02:57 serv python[18604]:     self.doRollover()
May 06 15:02:57 serv python[18604]:   File "/home/lucian/work/printer/Klipper/klippy/queuelogger.py", line 46, in doRollover
May 06 15:02:57 serv python[18604]:     logging.handlers.TimedRotatingFileHandler.doRollover(self)
May 06 15:02:57 serv python[18604]:   File "/usr/lib/python2.7/logging/handlers.py", line 350, in doRollover
May 06 15:02:57 serv python[18604]:     os.rename(self.baseFilename, dfn)
May 06 15:02:57 serv python[18604]: OSError: [Errno 13] Permission denied
May 06 15:02:57 serv python[18604]: Logged from file klippy.py, line 139

sniperlucian, 2019-05-07:
this is odd:

/etc/default/klipper  has included

`KLIPPY_ARGS = ....   -l /tmp/klippy.log`

however klippy logs to /var/log/klippy.log

I just had upgraded to Mint 19 and had to reinstall klippy due to broken VENV.

KevinOConnor, 2019-05-09:
The log indicates Klipper was started with `-l /var/log/klippy.log`.  It looks like you're doing a custom install, so something to track down in the install scripts.

-Kevin

sniperlucian, 2019-05-10:
yeah you right, found it.

`/var/log/klippy.log`

is the default path in the ubuntu18.04 install script.

```
  # Step 3: Install startup script
install_script()
{
# Create systemd service file
KLIPPER_LOG=/var/log/klippy.log
```

sniperlucian, 2019-05-11:
for reference:

install scipt for ubuntu 18.04 installs the file

`/etc/systemd/system/klipper.service`

not the default in

`/etc/default/klipper `

with reference of log file position to

`/var/log/klippy.log
`
which leads to trouble because missing rights to make a logfile rollover from python.

KevinOConnor, 2019-05-16:
Ah, that is odd.  @nathantsoi - do you know why the ubuntu script uses /var/log/klippy.log instead of /tmp/klippy.log?

-Kevin

nathantsoi, 2019-05-17:
hum... i do not. i guess i put it there b/c that's where logs usually go. didn't realize this was getting rolled in python. i did chown the file, but i guess this belongs in a sub-folder owned by the service user/group, or just in `/tmp`. def. feel free to move it!

sniperlucian, 2019-05-18:
I would actually prefer to have it in /var/log - as it is more linux specific. 

I think chown is not enough, since python tries to generate a new file. Might work with subfolder with user rights though.

KevinOConnor, 2019-05-21:
FYI, I updated the ubuntu install script to use /tmp/klippy.log (as that's what all the documentation uses) - commit 7c3ee0f1.

-Kevin

closed 2019-05-25 by KevinOConnor