Klipper3d/klipper #1616: MCU 'mcu' shutdown: ADC out of range. Nozzle (cold) temp stuck at 288c - 293c

black-steve, 2019-05-07:
I have no idea what happened but all of a sudden after installing a new nozzle on my Ender 3 klipper displayed a nozzle temp of like 292c. The screen and octoprint still display the same temperature but it's stone cold. Then i keep getting the following error: 

MCU 'mcu' shutdown: ADC out of range
This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
Klipper state: Not ready

I've replaced the thermistor, heater cartridge and restart the firmware multiple times but it doesn't change anything. When i try to change temp to 200c it immediately kicks back the same error again. I changed the max_temp in the config file to 300c to remove the error but the nozzle temp is still stuck around 288c - 292c. After this config change i am able to increase the bed temp and the bed temp is reporting correctly.  

Also for the life of me i can't figure out how to download the klippy.log so i can attach it. I putty into the pi and i can access the /tmp but it tells me that /tmp/klippy.log doesnt exist. I'm trying to at least determine if this is software or hardware issue. 

Side note. After changing the max_temp to 300c i am able to control X/Z/Y axis on the printer as well as adjust the bed temp. I'm simply stuck with the nozzle remaining stone cold and not able to increase or decrease the temperature.

black-steve, 2019-05-07:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3150815/klippy.log)

AxMod3DPrint, 2019-05-07:
> I guess you measured the resistance of your thermistor on the end of the
> wire already?

Check your hot end thermistor. A constant high or negative temperature is a
sign of a short.

black-steve, 2019-05-07:
I'm fairly new to this stuff. I don't know how to test the resistance. All I've done is replace the thermistor and the heater cartridge with a new one but it's still reading a constant high temperature even though the heater cartridge is still cold. I plugged the same replacement thermistor into my 2nd ender 3 to see if the same issue would move but my other printer continued to report correctly which tells me the issue wasn't the thermistor.

KevinOConnor, 2019-05-09:
It's hard to say what the hardware problem is, but it definitely appears to be an issue with the hardware.  I'd triple check the thermistor wires and perhaps try wiring it to a different plug on the board.

-Kevin

Paulbsh, 2019-05-10:
Check the thermistor pins on the board for a short. A multimeter will help

AxMod3DPrint, 2019-05-10:
If you have changed the thermistor, have you checked you have the correct
thermistor set in your printer configuration?

black-steve, 2019-05-19:
> If you have changed the thermistor, have you checked you have the correct thermistor set in your printer configuration?
> […](#)

Are you referring to the voltage of the thermistor? I know i plugged it into my other Ender 3 that is working correctly and it worked properly. I'm not sure what setting i am looking for to adjust this.

black-steve, 2019-05-19:
> It's hard to say what the hardware problem is, but it definitely appears to be an issue with the hardware. I'd triple check the thermistor wires and perhaps try wiring it to a different plug on the board.
> 
> -Kevin

Isn't each plug set for a specific purpose? or are they all universal?

black-steve, 2019-05-19:
> It's hard to say what the hardware problem is, but it definitely appears to be an issue with the hardware. I'd triple check the thermistor wires and perhaps try wiring it to a different plug on the board.
> 
> -Kevin

Also when i completely unplug the thermistor it still reports the same temp on the screen.

black-steve, 2019-05-21:
My multimeter finally came in the mail. Using this thing is near beyond my depth outside of testing electrical sockets. 

But here are the readings i am seeing on different rests.

I set the mutlimeter to resistance mode. i touched the thermistor wires plugged into the printer (turned on) and it gave me a consistent 7.5 reading. I unplugged the thermistor and touched the multimeter to the pins on the board and it also gave me a 7.5 readon.

I touched the end of the thermistor wires unplugged and it gave me a 121 reading. I tried it on another printer and extra thermistors i have on hand and they all gave me readings around 120 range.

Now i wanted to see if i'd get consistent results touching the pins to my ender 3 printer that is working flawlessly. It gave me straight 0's, and when i moved the multimeter around on the pins eventually i got it to show like 0.03. with the thermistor plugged in or not i get the same 0.00 result on the screen of the multimeter. I have no idea what this means but i'd assume i'd get 0's on my printer that ISN'T working, not the one that is working..... thoughts? Any thing i should test with the multimeter?

black-steve, 2019-05-21:
The thermistor pins on the mainboard shorted out. I am replacing the board.

closed 2019-05-21 by black-steve