Klipper3d/klipper #1619: Question about Delta calibration

forerunnert, 2019-05-08:
I am calibrating my delta printer using the basic (using paper test) and enhanced delta calibration procedure outlined in Delta_Calibrate.md.
Before I started I printed a 20mm cube and measured 20.05 mm in X and 20.09 in Y.
After one cycle of basic and enhanced calibration I measured about ~20.30 mm in X and Y. After the second cycle I measured ~20.15 mm in X and Y.

Two questions:
- can I expect the accuracy to keep increasing the more calibration cycles are run?
- the procedure mentions in the additional notes "the distance between any two pillars should be around 74mm and the width of every pillar should be around 9mm". I am getting ~10mm and ~75mm respectively.
It also states "Specifically, the goal is for the distance between any two pillars minus the width of one of the pillars to be exactly 65mm" which is what I am seeing (75-10 vs 74-9). Are the absolute measurements relevant or does only the distance between pillars count? I am confused by the mm difference.

Thanks,
Michel

KevinOConnor, 2019-05-09:
That sounds like "filament swell" - the printer is calibrated correctly, but the filament expands a little when it cools.  (That, or the extruder step_distance needs to be calibrated.)  So, you've successfully calibrated the delta kinematic components, but you may need to calibrate the filament.

-Kevin

forerunnert, 2019-05-10:
That sounds plausible: I didn't spend much time calibrating the filament. I can also double check calibration with an indicator.
Thanks!

closed 2019-05-10 by forerunnert