Klipper3d/klipper #1622: Anycubic Predator - Delta ARM STM32F103ZET6

peterlpop, 2019-05-09:
Is this chipset supported - I saw in the "make" menu STM32F1 ... would I be able to use Klipper with this printer. While I don't understand ARM CPU (registers etc) details, I am getting quite comfortable  with the Klipper config files. 

1. If I simply "make" the firmware like I would with an ATMEGA (if that this specific STM32F1 is supported), configuring the config file should be straight forward (with some trial and error)?
2. How do I flash back to the original firmware if needed (I would hope that maybe somebody here has some experience)
3. Likely it's high-res touch panel won't work (I hope I am wrong)... is there a way to control/mirror the original Klipper firmware through OctoKlipper/OctoPi? - the Z offset while printing the raft/skirt is something I would put on the first screen/main screen of Klipper LCD (rather than Print Speed)
[image]

Obviously, any work I do on my by Predator printer I would share back with you all.

KevinOConnor, 2019-05-09:
Klipper has support for the micro-controller, but it is unknown if it will run on that particular board.  Some vendors have shipped stm32f103 chips on boards with a locked down bootloader.  You'll need bootloader access to install (and revert) Klipper.  You'll also need to know the pin schematic in order to define the config file.

-Kevin

peterlpop, 2019-05-10:
So if I understand correctly, if I try to flash it with Klipper, if it
works than that means I have access to the boot-loader? Or I need to flash
a boot-loader first like I had to do with Ender (though a different way
since this is not Aruduino).

And if I can't flash it, then I likely can't break it to where I can't
flash its own firmware back?

As for the diagram, I would hope the factory would provide the info?
_______________________
Peter Lucian Pop
Google Voice, 801-810-8129
peterlpop@gmail.com

KevinOConnor, 2019-05-14:
Alas, I don't know the answer to any of those questions.  Someone with the board would have to work through the above and determine what is necessary.

-Kevin

KevinOConnor, 2019-05-25:
I'm going to close this issue as it looks like the conversation has concluded.  FYI, I've seen reports that Klipper has been able to run on the "bigtreetech skr mini" which is also a stm32f103 board.  So, there's some hope that it will run on similar boards.

-Kevin

closed 2019-05-25 by KevinOConnor