Klipper3d/klipper #1623: endstop_phase - how to configure correctly?

StefanMeGit, 2019-05-09:
Hi there

I try to find out how the ENDSTOP PHASE section in the config should looks like.

I started to add `[endstop_phase]` into my config and go multiple times for G28 and `ENDSTOP_PHASE_CALIBRATE` . Then I saved everything with `ENDSTOP_PHASE_CALIBRATE STEPPER=stepper_z` and `SAVE_CONFIG` .

After that I found this section in the examples:

```
# Stepper phase adjusted endstops. To use this feature, define a
# config section with an "endstop_phase" prefix followed by the name
# of the corresponding stepper config section (for example,
# "[endstop_phase stepper_z]"). This feature can improve the accuracy
# of endstop switches. Add a bare "[endstop_phase]" declaration to
# enable the ENDSTOP_PHASE_CALIBRATE command.
#[endstop_phase stepper_z]
#phases:
#   This specifies the number of phases of the given stepper motor
#   driver (which is the number of micro-steps multiplied by four).
#   This setting is automatically determined if one uses TMC2130,
#   TMC2208, TMC2224 or TMC2660 drivers with run-time configuration.
#   Otherwise, this parameter must be provided.
#endstop_accuracy: 0.200
#   Sets the expected accuracy (in mm) of the endstop. This represents
#   the maximum error distance the endstop may trigger (eg, if an
#   endstop may occasionally trigger 100um early or up to 100um late
#   then set this to 0.200 for 200um). The default is
#   phases*step_distance.
#endstop_phase:
#   This specifies the phase of the stepper motor driver to expect
#   when hitting the endstop. Only set this value if one is sure the
#   stepper motor driver is reset every time the mcu is reset. If this
#   is not set, then the stepper phase will be detected on the first
#   home and that phase will be used on all subsequent homes.
#endstop_align_zero: False
#   If true then the position_endstop of the axis will effectively be
#   modified so that the zero position for the axis occurs at a full
#   step on the stepper motor. (If used on the Z axis and the print
#   layer height is a multiple of a full step distance then every
#   layer will occur on a full step.) The default is False.
```
Do I have to configure this also? At the moment my config looks like this:

```
[endstop_phase]

[endstop_phase stepper_z]
endstop_align_zero: True

[endstop_phase stepper_z1]
endstop_align_zero: True

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [endstop_phase stepper_z]
#*# endstop_phase = 40
```

Thanks for help!

StefanMeGit, 2019-05-09:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3162151/klippy.log)

KevinOConnor, 2019-05-09:
It's not necessary to explicitly configure `[endstop_phase stepper_z]` if you've used `ENDSTOP_PHASE_CALIBRATE` and `SAVE_CONFIG`.  However, if you want to enable `endstop_align_zero` then you would configure it as you have above.  Note, though, that an endstop_phase does not apply to stepper_z1 (both stepper_z and stepper_z1 are synced together irrespective of any endstop_phase declaration).

-Kevin

StefanMeGit, 2019-05-11:
Thank u very much!

closed 2019-05-11 by StefanMeGit

changedsoul, 2020-09-28:
I am going to reopen this. I have been experimenting with the endstop phase stuff and I am a weee bit confused.
I am issuing a G28 for my Z axis and after wards running the "ENDSTOP_PHASE_CALIBRATE STEPPER=stepper_z", then going to the console and look at the output. I am seeing stuff that I don't understand, and I hope you can explain it to me.
```
// stepper_z: endstop_phase=31 (range 30 to 33)
// stepper_z: endstop_phase=32 (range 31 to 33)
// stepper_z: endstop_phase=31 (range 31 to 31)
// stepper_z: endstop_phase=32 (range 31 to 32)
// stepper_z: endstop_phase=32 (range 32 to 32)
// stepper_z: endstop_phase=28 (range 28 to 28)
```
I have changed nothing as far as a limit switch adjustment. I just ran the homing multiple times and then ran the calibrate and then checked the console. I got these results.
Can you explain what range is and why it keeps changing? Also, if this is changing all the time, does it mean my "endstop_accuracy" needs to be modified from default of its calculated value?

Here is my config.
[klipper.config.txt](https://github.com/KevinOConnor/klipper/files/5292994/klipper.config.txt)