Klipper3d/klipper #1624: [Feature Request] Physical Inputs ie: Buttons, Potentiometers

mental405, 2019-05-09:
It would be really nice to have physical button support beyond the few typically found on display controllers. These buttons could be tied to gcode macros. 

Example functionality for a digital input would be a button that you press and it executes a gcode macro that disables  or enables the motors if the printing state is idle.

Another example would be a pin that halts all movement and force moves the z axis up a predefined distance in the event that a power loss is detected from some sensor. I'm not sure with the scheduler this is possible but it could be cool.

Example functionality for an analog input would be a pin tied to a joystick(s) that jogs the x/y/z axis if the printing state is idle. Pressing the center of the joystick could home the axes. 
I am interested in making a USB cabled remote control that is home to an ATmega32u4 or at90usb1286, some controls, and a screen, using klipper's built in multi-mcu support to managed the addional pins.

JoaoFelipe, 2019-05-10:
If you use octoprint with klipper, you can use the GPIO of the RPi to add these buttons and create a plugin to send the gcode to klipper. 
I did something like that: https://github.com/JoaoFelipe/OctoPrint-Krestart
I have buttons to send "FIRMWARE_RESTART", "RESTART", and "M112" to the printer

On a side note, I would like to use these buttons to control the screen as well. For instance, add a button to open the Control submenu. Unfortunately, I couldn't find options/gcode for that.

mental405, 2019-05-10:
I had thought about the pi route but that seems like it would limit the number of GPIO pins available as well as the capabilities.

KevinOConnor, 2019-05-16:
If someone wishes to add an "extras" module that can run a g-code script on a button event, then that is great.  See https://www.klipper3d.org/Code_Overview.html#adding-a-host-module for details.

-Kevin

KevinOConnor, 2019-05-25:
I'm going to mark this as inactive as it doesn't appear anyone is actively working on the development for this.

-Kevin

closed 2019-05-25 by KevinOConnor

mental405, 2019-06-22:
This was mostly implemented with [#1745](https://github.com/KevinOConnor/klipper/pull/1745).