Klipper3d/klipper #1625: Move exceeds maximum extrusion (1.863mm^2 vs 0.640mm^2)

DioABD, 2019-05-10:
Hi, i get the "Move exceeds maximum extrusion" Error mid Print after 1 hour.

Log file attached. Slicer used: Simplify3D 4.1.1

Printed many files fine before. 

Terminal output while that occours:

Send: N82459 M84*29
Recv: ok
Send: N82460 M104 T0 S0*41
Recv: // Move exceeds maximum extrusion (1.863mm^2 vs 0.640mm^2)
Recv: // See the 'max_extrude_cross_section' config option for details
Recv: !! Move exceeds maximum extrusion (1.863mm^2 vs 0.640mm^2)
Send: N82461 M108*35
Send: N82462 M140 S0*111
Recv: ok
[klippy.log](https://github.com/KevinOConnor/klipper/files/3166444/klippy.log)

Can anybody help me fix this issue? 

Thank you very much
Best regards, Dio

DioABD, 2019-05-10:
It always happens on the same spot. Sliced few times. Updated Klipper and mcu. Nothing helps.

Issue occours always on bridging part when fan is turned to 100%. Right after the bridging when the fan goes back to lower speed the issue stops the print.

DioABD, 2019-05-10:
I have set max_extrude_cross_section to 50.
But it it still fails.  

New log attached.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3167640/klippy.log)

jakep82, 2019-05-10:
Your log does not show max_extrude_cross_section as a configured parameter.  Did you edit the wrong config file, or forget to restart the printer after editing?  

For reference this issue has come up over and over again, and in almost every case the slicer that created the bad gcode was S3D.  I've personally stopped using S3D most of the time because of this issue.  I don't like disabling safety features on my printer to accommodate bad software.

DioABD, 2019-05-10:
I will check the config. Maybe I forgot to restart. 

Yes maybe S3D is the only slicer that leads to this issue. But also Klipper is the only printer Firmware with this issue.

And when somebody pays a lot money on S3D and tunes all settings and everything and it works fine with all other printer firmwares, it is no option to change the slicer for me. It cost me as much as the printer. And it worked perfect for a lot of prints. 

And what does safety Feature mean in this case? The printer does nothing strange. It just prints 1cm bridging in the middle of the Heatbed. If that is recognized as a safety risk than the recognition is simply wrong. On Marlin the print finishes perfect. No risk at all. It is unnecessary to stop a long time print for nothing.

Safety feature means to me things like overheating or something that could break Hardware etc. But in this case there is no risk at all. I watched the printer exactly several times when I tried to print it. Not even near to any risk that would need an instant stop wasting the print.

So just pointing to S3D as the reason is a bit too easy I think. It should be at least an option available to disable that because there is  no real problem nor a real safety Issue. Should the May extrusion command in Extruder part of the config fix my issue in this case? Do i need to just write a number after the command? Or does it need something like „:“ between?

Thx for your help

jakep82, 2019-05-10:
I'm not going to explain why this safety feature exists since it's already been explained in other issues in this repository.  If you search for `move exceeds maximum extrusion` you'll find a lot of information. 

As for working around the issue, you've already found the solution, but your log doesn't match the information you've posted here regarding `max_extrude_cross_section`

DioABD, 2019-05-10:
Thank you for your response.

What line do i exactly need to put into the printer.cfg? Can you show an example? Maybe i have some wrong sign inserted or something. Also is the „Extruder“ Part correct? Or should it be the „Extruder 1“ part where my TMCs are configured?

Thx a lot for your help

jakep82, 2019-05-10:
Just copy/paste from the example config, add whatever number you're comfortable with (presumably 50 as you mentioned above), and remove the hashtag to uncomment the line.  This line should be in your `[extruder]` section as shown in the example.

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L107

DioABD, 2019-05-11:
Okay, i have added this line:
max_extrude_cross_section: 50.0
into the [Extruder] section and will retry.

I was unsure, because i have also sections like [tmc2208 extruder] which the example does not have.

DioABD, 2019-05-11:
It worked !!!! :D  

The print passed the part where it stopped 5 times before ruining the print. 
And guess what... It printed perfectly. No issues.

That proves for me that the "safety function" is not working correctly, it is overreacting. Also there was not extruded to much or anything else. It printed perfectly just as expected. Iam happy i could avoid this false function. Maybe it false detected the speed change on bridging or the difference in extruding amount while bridging.

Thank you very much for your support...

The dev team should have a look at this function. The amount of tickets on this show how much people run into this issue, while klipper is able to print the gcode fine. The tolerance should at least be higher.

Now i used the value 50 for this, which is maybe a bit too much to fix it. Which value would solve the issue in the case of: 1.863mm^2 vs 0.640mm^2  ?

How much mm2 would a value of 10 allow?

KevinOConnor, 2019-05-25:
I'm going to close this issue as it appears a solution was found for the original issue.

-Kevin

closed 2019-05-25 by KevinOConnor