Klipper3d/klipper #1626: No icons in customizable display

dystechnic, 2019-05-10:
Hi,
In an effort to fix my issue with multiple extruders and just one heater, [#1580](https://github.com/KevinOConnor/klipper/issues/1580), I tried creating a custom card menu.
In basic this works, but I am unable to get the icons for nozzle, heated bed, etc. on the display. According to example-menu.cfg a nozzle icon should appear if \x00 is used in the menu definitions.
But in my case, no icons are displayed.

I have a reprapwolrd full graphics display (st7920) and in the default menu the icons are shown okay. 

This is what I have configured in my printer.cfg:

`[display]
lcd_type = st7920
cs_pin = ar62
sclk_pin = ar76
sid_pin = ar75
encoder_pins = ^ar21, ^ar20
click_pin = ^!ar64

[menu __infodeck]
type = deck
name = infodeck
longpress_menu = __main
items = __infodeck_card_static

[menu __infodeck_card_static]
type = card
name = Card infodeck
content:
    "\x00{0:3s}\x04\x7e{1:3s}\x04   {2}"
    "\x01{3:3s}\x04\x7e{4:3s}\x04   \xa5{8}"
    "\x07{6}  \x02{5} {7}"
    "{9}"
items = 
	__card_hotend0_current
	__card_hotend0_target
	__card_zpos
	__card_hotbed_current
	__card_hotbed_target
	__card_frpeed
	__card_fnspeed
	5,__card_prt_time, __card_usb_progress, __card_sdcard_progress
	__card_status
	__card_msg,__card_xpos|__card_ypos|__card_epos
`
Is what I am trying possible with the menu system? Am I overlooking something in the menu config?

Thanx,
[klippy.log](https://github.com/KevinOConnor/klipper/files/3167463/klippy.log)

mcmatrix, 2019-05-11:
It’s not supported in current menu.
It will be possible in #1518

dystechnic, 2019-05-11:
Oke, thanx.
I found that commenting out a line in display.py gives me the result I was after. 
So I can wait for the changes in the menu system to do it as it was ment to be done.

KevinOConnor, 2019-05-25:
I'm going to close this issue as it looks like the original question has been answered (it's not currently supported).

-Kevin

closed 2019-05-25 by KevinOConnor