Klipper3d/klipper #1627: Bondtech Sounds Rough

totalitarian, 2019-05-10:
I have a bondtech bmg fitted to my delta. It sounds rough as hell under clipper.

https://youtu.be/TL2iexAwQbc

 BTW the motor makes the same screeching even with the bondtech remove. For some reason the change in steps makes my motor very upset :)

Sorry not log as i've switched back to my previous extruder but I noticed a few people on facebook with the same issue. Config attached

[config.txt](https://github.com/KevinOConnor/klipper/files/3167529/config.txt)

jakep82, 2019-05-10:
It appears you're using a fairly high pressure advance value, which is common with bowden extruder setups.  That's the cause of the noise.  You can either reduce/disable pressure advance, or live with the noise.

totalitarian, 2019-06-04:
Still battling this one. This is what mine sounds like (not my video)  https://www.youtube.com/watch?v=isIyaAJv3lQ

It makes this sound even when the bmg is not connected. Acceleration is turned down to 600. I'm running TMC2208's in standard mode.

Here is my log

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3253825/klippy.zip)

jakep82, 2019-06-04:
You shouldn't run 2208 drivers in standalone.  This has been discussed many times and [documented.](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print)  They will shut down during a print.  In a best case scenario it only ruins your print.  In a worst case scenario you could ruin your hotend and possibly start a fire.

Regarding the noise, that is the result of using high pressure advance values (I see 0.7 in your log).  The noise is likely the motor missing steps as it tries to make very fast direction changes.  You can try using a larger motor running at a higher amperage, but that isn't guaranteed to work.  The best solutions are using a lower pressure advance value, or reducing the velocity and acceleration of your entire printer.

totalitarian, 2019-06-04:
I know some people have issues with 2208s but i've clocked up hundreds of hours with them with no issues with shutting down. I could never get them configured via uart. There is a thread https://github.com/KevinOConnor/klipper/issues/1588. I'm not sure pressure advance plays a part as i'm simply running G1 E-6.5000 F4200 from the terminal

jakep82, 2019-06-04:
If all you're trying to do is retract, then turn down your speed.  70mm/s is really fast for a geared extruder.  That would be the equivalent of 210mm/s with a non-geared extruder.  Even then, I still think you'll have a lot of issues with the pressure advance value you're using.

jakep82, 2019-06-04:
For reference I use 30mm/s for retractions with my Bondtech extruder.

jakep82, 2019-06-04:
Here is a more in depth answer regarding what's happening with your motor.  It's most common on the Z axis, but with a geared extruder running at higher speeds you'll have the same issue.

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-converted-my-config-from-marlin-and-the-xy-axes-work-fine-but-i-just-get-a-screeching-noise-when-homing-the-z-axis

totalitarian, 2019-06-04:
I guess that makes sense as I never had this issue with Marlin, linear advance and my BMG. Acceleration is only set to 600mm/s in klipper where I could get away with 1000mm/s in marlin

elcojacobs, 2019-06-04:
I'm using this config with a BMG clone and TMC2130 drivers. TMC2208 will be similar. My BMG is very silent now.

Stealtchop works great, but you'll have to increase the aggressiveness of the algorithm (PWM GRAD) and set the threshold very high so it never falls back to non-stealtchop mode.
The switch will cause a hiccup and print errors.

[tmc2130 extruder]
cs_pin: ar42
microsteps: 4
run_current: 0.5
hold_current: 0.5
driver_PWM_GRAD: 8
stealthchop_threshold: 5000

[extruder]
pressure_advance: 0.45
pressure_advance_lookahead_time: 0.02

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 500
max_z_velocity: 10
max_z_accel: 50
square_corner_velocity: 5

More here:
https://github.com/KevinOConnor/klipper/issues/1363

KevinOConnor, 2019-06-18:
Okay, I'm going to close this issue as it looks like the original question was answered.

-Kevin

closed 2019-06-18 by KevinOConnor