Klipper3d/klipper #1630: Firmware crash on LCD input

Nibb31, 2019-05-12:
Hi,
I'm using a Full Graphic Smart Controller - 12864 LCD with rotary knob on an MKS Gen 1.4 controller board (https://www.roboter-bausatz.de/198/lcd-12864-reprap-full-graphic-smart-controller-fuer-ramps-1.4-mit-sd-card-reader). Octoprint and Klipper are latest updates running on an RPI 3A+. 
When I started using Klipper a few months ago, this worked fine for navigating the menus, except that pressing the STOP button would cause the MCU firmware to freeze. Not a big deal for me.
However, since a couple of updates, turning or pressing the knob also causes Kipper to freeze, which makes the LCD menu unusable.
I have attached the log file.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3169949/klippy.log)

gdachs, 2019-05-12:
Why not first fix the error you have in klippy.log? Maybe that even fix your problems with freezing.

Nibb31, 2019-05-12:
I remove the virtual_sdcard and it does fix the problem.
There is still the issue with the STOP button crashing the firmware.

mcmatrix, 2019-05-13:
The STOP button is calling "Emergency Stop".
It will shut down the the klipper. 
It should be the same as calling `M112`

closed 2019-05-14 by Nibb31