Klipper3d/klipper #1631: Shared Heater different Moving Direction

AniInABox, 2019-05-12:
Hello,

someonr has an idea how to solve this?
I want to configure a shared Heater with the same Motor but different Moving Direction like the original Extruder.

Any Suggestions?

https://www.thingiverse.com/thing:1643906

KevinOConnor, 2019-05-16:
Support for that would require coding changes.  Take a look at klippy/kinematics/extruder.py if interested.

-Kevin

AniInABox, 2019-05-19:
Any Suggestions how to start?
Since we need 1 Motor for 2 Directions, we had to know both positions correct?

I will check the extruder.py...

KevinOConnor, 2019-06-09:
I'm going to mark this as inactive as it looks like the conversation has concluded for now.  The best hints we have is the developer documentation at https://www.klipper3d.org/Overview.html .

-Kevin

closed 2019-06-09 by KevinOConnor

KevinOConnor, 2022-02-10:
FYI, this may now be possible with the merge of PR #5210 .

-Kevin