Klipper3d/klipper #1632: issue installing on ender 3

NicoSoftware, 2019-05-12:
I got to the point Where I had to make a flash of the USB Port:
```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x65
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x68
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x6f
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x55
```
Having an ender 3, with bltouch/v6 hotend and not a correct marlin.
I've made mechanical changes but didn't upgrade marlin to them, is this a problem?`
Thanks in advantage

guerillaengineer, 2019-05-13:
Have you installed a bootloader onto the ender 3 board?

NicoSoftware, 2019-05-13:
> Have you installed a bootloader onto the ender 3 board? Just a shot in the dark question.

Yes, I've installed marlin over USB before.
I'll post them later, when I get home from work.
Thanks alot in advantage!

NicoSoftware, 2019-05-13:
> EDIT: I am commenting so I can see your logs. When you post them `please reply to me with a copy` and I would love to help you.

/tmp/klippy.log:
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-457-g777013d'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Start printer at Mon May 13 17:06:06 2019 (1557767166.3 8.1)
Unable to open config file /home/pi/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 96, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 133, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 119, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 213, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 102, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/printer.cfg
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-457-g777013d'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Start printer at Mon May 13 17:11:15 2019 (1557767475.4 309.6)
Unable to open config file /home/pi/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 96, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 133, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 119, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 213, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 102, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/printer.cfg

NicoSoftware, 2019-05-13:
Did a reinstall and I think thats pretty much the same error:
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-457-g777013d'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Start printer at Mon May 13 17:06:06 2019 (1557767166.3 8.1)
Unable to open config file /home/pi/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 96, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 133, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 119, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 213, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 102, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/printer.cfg
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-457-g777013d'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Start printer at Mon May 13 17:11:15 2019 (1557767475.4 309.6)
Unable to open config file /home/pi/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 96, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 133, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 119, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 213, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 102, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/printer.cfg
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-457-g777013d'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Building C code module c_helper.so
Start printer at Mon May 13 17:49:43 2019 (1557769783.4 2617.6)
Unable to open config file /home/pi/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 96, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 133, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 119, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 213, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 102, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/printer.cfg

NicoSoftware, 2019-05-13:
After installing the klipper addon on the octopi GUI and just simply added a printer.cfg through the klipper settings I get this error:
```
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-457-g777013d'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Start printer at Mon May 13 18:45:50 2019 (1557773150.4 8.1)
===== Config file =====
[stepper_x]
step_pin = PD7
dir_pin = !PC5
enable_pin = !PD6
step_distance = .0125
endstop_pin = ^PC2
position_endstop = 0
position_max = 235
homing_speed = 50

[stepper_y]
step_pin = PC6
dir_pin = !PC7
enable_pin = !PD6
step_distance = .0125
endstop_pin = ^PC3
position_endstop = 0
position_max = 235
homing_speed = 50

[stepper_z]
step_pin = PB3
dir_pin = PB2
enable_pin = !PA5
step_distance = .0025
endstop_pin = ^PC4
position_endstop = 0.0
position_max = 250

[extruder]
max_extrude_only_distance = 100.0
step_pin = PB1
dir_pin = !PB0
enable_pin = !PD6
step_distance = 0.010526
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PD5
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PA7
control = pid
pid_kp = 21.527
pid_ki = 1.063
pid_kd = 108.982
min_temp = 0
max_temp = 250

[heater_bed]
heater_pin = PD4
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PA6
control = pid
pid_kp = 54.027
pid_ki = 0.770
pid_kd = 948.182
min_temp = 0
max_temp = 130

[fan]
pin = PB4

[mcu]
serial = /dev/ttyUSB0

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100

[display]
lcd_type = st7920
cs_pin = PA3
sclk_pin = PA1
sid_pin = PC1
encoder_pins = ^PD2, ^PD3
click_pin = ^!PC0
=======================
Extruder max_extrude_ratio=0.266081
Starting serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.

Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.

Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.

Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.

Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Got error -1 in read: (11)Resource temporarily unavailable
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
```

jakep82, 2019-05-13:
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

NicoSoftware, 2019-05-13:
> https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

Yeah I fixed this like 3 minutes ago haha but thank you

jakep82, 2019-05-13:
Post a new log file.  And by that I mean attach the file.  Please don't copy/paste a wall of text.

NicoSoftware, 2019-05-13:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3174550/klippy.log)

jakep82, 2019-05-13:
Have you successfully flashed the board?

NicoSoftware, 2019-05-13:
> Have you successfully flashed the board?

not via CLI, I think it worked over octoprint, but I can't tell

jakep82, 2019-05-13:
You can't flash the board via Octoprint.  Please read the installation document and make sure you've followed all the instructions.

https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#building-and-flashing-the-micro-controller

NicoSoftware, 2019-05-13:
Yeah it works now, I don't know why he didn't wanted to flash it. But it works now.
Raspberry pi's are so strange sometimes. anyway thank you!

closed 2019-05-13 by NicoSoftware