Klipper3d/klipper #1633: BEd_Mesh_Leveling

cima37, 2019-05-12:
Hello!

I have a Hypecube EVO with piezo Orion z endstop and this triggers at Bed Mesh Leveling!
Is it possible to insert a break at BED_MESH_Leveling before the z-axis moves? So that the piezo Orion does not respond?

cima37, 2019-05-12:
Here's the log I forgot!
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3170512/klippy.log.zip)

emailcraigs, 2019-05-13:
Can the mesh points be modified in the config file to adjust the entire mesh by a set amount or will this somehow cause issues with implementing the mesh?

cima37, 2019-05-13:
I can set the number of points in the x-axis and the y-axis and how often per point is measured but nothing else!

Arksine, 2019-05-16:
@emailcraigs I suggest applying a gcode offset using SET_GCODE_OFFSET rather than attempting to offset or modify the mesh.

@cima37 I'm not exactly sure what you are asking.  You have a piezo style probe, and its triggering before the nozzle touches the bed?  Of you need some kind of delay before probing you can add it to the probe's activate gcode.

cima37, 2019-05-16:
@Arksine jest i need a delay! How can i add a delay in the probe?

theopensourcerer, 2019-05-16:
https://reprap.org/wiki/G-code#G4:_Dwell

cima37, 2019-05-16:
Thank you! It works!

closed 2019-05-16 by cima37