Klipper3d/klipper #1635: DUET Wifi flahs fails

VirtualAI, 2019-05-12:
Hello,
i just installed klipper on my pi and wanted to flash my DUET Wifi.
After running make and creating the confi file, i was running
sudo make flash FLASH_DEVICE=/dev/serial/by-id/usb-Duet3D_Duet-if00
which results in 

  Flashing out/klipper.bin to /dev/serial/by-id/usb-Duet3D_Duet-if00
Entering bootloader on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.2:1.0
Device reconnect on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.2:1.0
lib/bossac/bin/bossac -U -p /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.2:1.0 -e -w out/klipper.bin -v

No device found on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.2:1.0
Failed to flash to /dev/serial/by-id/usb-Duet3D_Duet-if00: Error running bossac
src/atsam/Makefile:65: recipe for target 'flash' failed

also 
ls /dev/serial/by-id/*
leads to
/dev/serial/by-id/usb-Duet3D_Duet-if00

i don't know any further.
Best regards.

VirtualAI, 2019-05-13:
Installation error provided at top,  no running log

mweber99, 2019-05-13:
Hello, it looks you and I are at the same junction trying to setup a Duet 2 Wifi. I was able to get the firmware to load by jumping the Erase pins on the board and pressing the erase button. After doing this and taking the erase jumper off, the firmware loads. However, I am not able to get Klipper on the PI to talk to the controller. The serial ports seems to be messed up after doing this, I do not have an entry for serial, only serial1, no entry for /dev/ttyACM0. I only have Serial1 which points to /dev/ttyAMA0. I've tried both and I get 

So, proceed with caution, I don't know if doing the reset caused more problems than it fixed.

mweber99, 2019-05-13:
Here is my log for what its worth...

[klippy.log](https://github.com/KevinOConnor/klipper/files/3174485/klippy.log)

VirtualAI, 2019-05-13:
Hello, thanks.
I managed to install firmware using a tool named Bossa (Duet Wifi connected to Windows, flashing .bin)
No i resieve new error while parsing the config file.
Config file and log file attached
[klippy.log](https://github.com/KevinOConnor/klipper/files/3174951/klippy.log)

[printer.txt](https://github.com/KevinOConnor/klipper/files/3174953/printer.txt)

KevinOConnor, 2019-05-16:
The log indicates that the micro-controller code is failing during the configuration process.  Best guess is that the firmware was not successfully flashed to the micro-controller.  It's unclear why you are unable to enter the firmware with the standard `make flash` command.  For the very first time flashing Klipper, it may be necessary to manually enter the bootloader on the Duet by holding the erase button down while pressing the reset button.

-Kevin

KevinOConnor, 2019-05-22:
FYI, a similar problem was reported and resolved in issue #1640.

-Kevin

closed 2019-06-02 by KevinOConnor