Klipper3d/klipper #1636: Installation problems

patoune08, 2019-05-12:
Hello,
It's been three days since I tried to install Klipper on a Tronxy XY-2 printer, motherboard with STM32F103ZE processor, but I can not do it.
I attach the Klipper.log file as requested, I added a Klipper2.log file that contains what was displayed in the terminal during installation.
I hope that someone could help me.
Thank you in advance.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3170676/klippy.log)
[klippy2.log](https://github.com/KevinOConnor/klipper/files/3170677/klippy2.log)

guerillaengineer, 2019-05-13:
From what I am catching it is not finding the printer.cfg where it needs (/home/pi/printer.cfg)
I would go through the bottom area of the installation guide titled "Configuring Klipper" again.

I don't know if this is connected to your config issue but the second log doesn't show the pi connecting to your mcu. I would give flashing the board another shot.

In the log it showed that you tried to input the direct commands for flashing, you need to substitute the device serial port with your own by running `ls /dev/serial/by-id/*` and substituting that into the commands.

I believe this is my first time commenting advice on github, so if I've overlooked something asinine I apologize!

gdachs, 2019-05-14:
Flashing of the bootloader failed. Did you check that /dev/ttyAMA0stm32flash exists?

KevinOConnor, 2019-05-15:
This issue looks similar to issue #1622.  A number of STM32F103ZE boards use an unusual bootloader configuration.  It should be possible to install Klipper on these boards, but the steps necessary to accomplish it are not known.  So, someone with the board will need to figure out what steps are necessary and then provide that information.

-Kevin

patoune08, 2019-05-15:
Good evening,
First of all thank you for your answer.
I've attached what's going on using "make serialflash FLASH_DEVICE = / dev / serial / by-id / usb-1a86_USB2.0-Serial-if00-port0" following the result of "ls / dev / serial / by-id / * ".
I do not have a file "ttyAMA0stm32flash" in / dev /
[klipper.log](https://github.com/KevinOConnor/klipper/files/3184546/klipper.log)

yo2mpn, 2019-05-16:
@patoune08 
Try this script to list al serial ports ,rename .txt to .sh . 
Result should look like this :

/dev/input/mouse0 - Microsoft_Microsoft_R__Compact_Optical_Mouse
/dev/input/event0 - Microsoft_Microsoft_R__Compact_Optical_Mouse
/dev/input/event1 - USB_USB_Keykoard
/dev/input/event2 - USB_USB_Keykoard
/dev/ttyACM0 - LeafLabs_Maple

[portlist.txt](https://github.com/KevinOConnor/klipper/files/3185488/portlist.txt)

KevinOConnor, 2019-05-25:
I'm going to close this issue as it does not appear anyone knows the answer to the original question.  Someone with the board will need to figure out the installation steps for it.

-Kevin

closed 2019-05-25 by KevinOConnor

patoune08, 2019-06-09:
Hello,
I come back with new:
I attach the loader of my card if someone can do something because I really want to be able to install Klipper.
Thank you.
[loader_chuanxingyuan.zip](https://github.com/KevinOConnor/klipper/files/3269573/loader_chuanxingyuan.zip)