Klipper3d/klipper #1639: [Request] Power Off when exceeding max temp

Blankomc, 2019-05-13:
Hello, I am trying to configure the firmware for when the maximum temperature defined in extruder or hot bed the printer is turned off. It is a security system that implements since a relay failed and left the hot bed fed reaching a temperature of 280 ° and almost ignite fire. The solution is to cut off the power and turn off the printer completely. Someone to help me or tell me if it is possible in klipper? Thanks.

gdachs, 2019-05-14:
I use a heated bed that is powered by 230V via a SSR. Before that SSR I have an mechanical relay. On start of Klipper that relay is switched on. On all kinds of errors that lets Klipper go into its shutdown state it is powered off and the SSR gets no voltage anymore.
This is all I need to do:
`[output_pin break_circuit_relay]
    #servo pin
    pin:            ar6
    value:          1
    shutdown_value: 0`

Blankomc, 2019-05-14:
the option you give me is not worth it, having the pin configured with M81 does not let me assign it to another option. in my case the hot bed is 12v and is through a mosfet. I need the printer to turn off completely. The time I had the problem was that the relay I had was still active sending the relay to cut the tension so I decided that the printer should be turned off completely.

Hywelmartin, 2019-05-14:
@gdachs:s suggestion could be realized for the whole printer.. but it will be hard to turn on if you don't power the board from usb..or you could have a spring loaded button to start it with(that bypasses the relay)..

gdachs, 2019-05-14:
@Hywelmartin yes, that was what I had in mind. But it is not worth it. He is hard to understand, unfriendly, and I don't see what my idea has to do with M81 (ATX Power Off).

Hywelmartin, 2019-05-14:
@gdachs  the relay-thingy is safer....

Blankomc, 2019-05-16:
All the machines I have have a push button to turn on, after about 5 seconds that the plate starts the relay activates and the button can be released. The level of security that it gives me is quite high since I print a lot during the night.

KevinOConnor, 2019-05-16:
It's not clear to me what is being reported or requested here.  The Klipper micro-controller will enter a shutdown state if the temperature goes out of the min_temp/max_temp range defined in the config file.  So, configure the appropriate temperature settings in the heater config sections.  If you want the micro-controller to also enable or disable a relay on a shutdown state, then you should configure the pin controlling the relay in the config file and specify a `shutdown_value` that triggers the given action.  See the example provided by @gdachs above.  Test the results by issuing an M112 command (as described in https://www.klipper3d.org/Config_checks.html ).

-Kevin

Blankomc, 2019-05-16:
What I want to do is simple, when the temperature exceeds the set margins at the same time that for printing I deactivate the pin of the relay that I use with the M81 and turn off the printer.

KevinOConnor, 2019-05-16:
Then set the appropriate `shutdown_value` on the pin that you've configured M81 to use.

-Kevin

KevinOConnor, 2019-05-25:
I'm going to close this issue as it appears the original question was answered (use a shutdown_value in the config).

-Kevin

closed 2019-05-25 by KevinOConnor