Klipper3d/klipper #1640: Duet Wifi 2 will not connect

mweber99, 2019-05-14:
Hello,
I'm trying to get my Duet 2 Wifi hooked up to Klipper. I am able to do the flash using Bossa on Windows but when I hook the Duet to the Raspberry Pi, it will not connect. 

Enclosed is the log and a screen shot showing the USB port configuration on the PI.

Thanks!

[klippy.log](https://github.com/KevinOConnor/klipper/files/3179921/klippy.log)
[image]

VirtualAI, 2019-05-14:
Hello,
i have compareable problem.
Error Log:
Start printer at Tue May 14 23:52:04 2019 (1557874324.5 350.7)
===== Config file =====
[stepper_x]
step_pin = PD6
dir_pin = PD11
enable_pin = !PC6
step_distance = .0125
endstop_pin = ^PC14
position_endstop = 0
position_max = 250

[tmc2660 stepper_x]
cs_pin = PD14
spi_bus = usart1
microsteps = 16
interpolate = True
run_current = 1.000
sense_resistor = 0.051
idle_current_percent = 20

[stepper_y]
step_pin = PD7
dir_pin = !PD12
enable_pin = !PC6
step_distance = .0125
endstop_pin = ^PA2
position_endstop = 0
position_max = 210

[tmc2660 stepper_y]
cs_pin = PC9
spi_bus = usart1
microsteps = 16
interpolate = True
run_current = 1.000
sense_resistor = 0.051
idle_current_percent = 20

[stepper_z]
step_pin = PD8
dir_pin = PD13
enable_pin = !PC6
step_distance = .0025
endstop_pin = ^PD29
position_endstop = 0.5
position_max = 200

[tmc2660 stepper_z]
cs_pin = PC10
spi_bus = usart1
microsteps = 16
interpolate = True
run_current = 1.000
sense_resistor = 0.051

[extruder0]
step_pin = PD5
dir_pin = PA1
enable_pin = !PC6
step_distance = .002
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = !PA20
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PC15
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
min_temp = 0
max_temp = 250

[tmc2660 extruder0]
cs_pin = PC17
spi_bus = usart1
microsteps = 16
interpolate = True
run_current = 1.000
sense_resistor = 0.051

[heater_bed]
heater_pin = !PA19
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PC13
control = watermark
min_temp = 0
max_temp = 130

[fan]
pin = PC23

[heater_fan nozzle_cooling_fan]
pin = PC26
heater = extruder0
heater_temp = 45
fan_speed = 1.0

[mcu]
serial = /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00
restart_method = command

[sx1509 duex]
i2c_address = 62

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100

[output_pin GPIO1]
pin = sx1509_duex:PIN_11
pwm = False
value = 1
=======================
Extruder max_extrude_ratio=0.266081
Starting serial connect
Loaded MCU 'mcu' 77 commands (v0.7.0-457-g777013d-20190514_233921-octopi / gcc: (15:5.4.1+svn241155-1) 5.4.1 20160919 binutils: (2.27-9+9) 2.27)
MCU 'mcu' config: BUS_PINS_twi1=PB5,PB4 BUS_PINS_twi0=PA4,PA3 BUS_PINS_spi=PA12,PA13,PA14 SOFT_PWM_MAX=256 STEP_DELAY=2 BUS_PINS_usart0=PB0,PB1,PB13 ADC_MAX=4095 BUS_PINS_usart1=PA21,PA22,PA23 PWM_MAX=255 CLOCK_FREQ=120000000 RESERVE_PINS_USB=PB10,PB11 MCU=sam4e8e STATS_SUMSQ_BASE=256
Sending MCU 'mcu' printer configuration...
Got error 0 in read: (0)Success

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.

Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 137, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 640, in _connect
    self._check_config()
  File "/home/pi/klipper/klippy/mcu.py", line 586, in _check_config
    config_params = self._send_get_config()
  File "/home/pi/klipper/klippy/mcu.py", line 570, in _send_get_config
    config_params = get_config_cmd.send_with_response(response='config')
  File "/home/pi/klipper/klippy/serialhdl.py", line 188, in send_with_response
    return src.get_response()
  File "/home/pi/klipper/klippy/serialhdl.py", line 222, in get_response
    raise error("Timeout on wait for '%s' response" % (self.name,))
error: Timeout on wait for 'config' response
Internal error during connect: Timeout on wait for 'config' response

mweber99, 2019-05-16:
Some further information. I was able to get the Duet firmware to load using the fallback 3 approach listed on the Duet setup website. https://duet3d.dozuki.com/Wiki/Installing_and_Updating_Firmware

The key was ensuring the Boot to Flash option is set so that after flashing the Bossa port is not shown. 

I’ve tried flashing Klipper using the approach documented on the Klipper site and I’ve tried  with the same options that worked with the duet firmware, the flash runs correctly but after a reboot it still only shows the Bossa port or sometimes no port at all. I’ve tried every possible approach and nothing works. 

Can anyone help me understand what I’m doing wrong as others seem I not have this problem. 

Thank You.

KevinOConnor, 2019-05-16:
I'm not sure what the current state of this issue is.

One should be able to flash the Duet2 by running the `make flash` command described at https://www.klipper3d.org/Installation.html#building-and-flashing-the-micro-controller .  If that is not working, please cut-and-paste the output from the `ls /dev/serial/by-id/*` and `make flash` command here.

-Kevin

mweber99, 2019-05-17:
Hi Kevin,
Thank you for the reply. I was able to load the Duet firmware to the board with no issues. 

I started by using the jumper to erase the Duet firmware and then followed your directions, what follows is the log. 

pi@ridingrasp:~/klipper $ lsusb
Bus 001 Device 020: ID 03eb:6124 Atmel Corp. at91sam SAMBA bootloader
Bus 001 Device 006: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@ridingrasp:~/klipper $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-03eb_6124-if00
pi@ridingrasp:~/klipper $ make menuconfig
scripts/kconfig/mconf /home/pi/klipper/src/Kconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

pi@ridingrasp:~/klipper $ sudo service klipper stop
pi@ridingrasp:~/klipper $ make flash FLASH_DEVICE=/dev/serial/by-id/usb-03eb_6124-if00
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/spi_software.o
  Compiling out/src/atsam/main.o
  Compiling out/src/atsam/gpio.o
  Compiling out/src/atsam/i2c.o
  Compiling out/src/atsam/spi.o
  Compiling out/src/atsam/hard_pwm.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/alloc.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/atsam/serial.o
  Compiling out/src/generic/serial_irq.o
  Compiling out/src/atsam/sam4e_afec.o
  Compiling out/src/atsam/sam4_cache.o
  Compiling out/src/../lib/sam4e/gcc/system_sam4e.o
  Compiling out/src/../lib/sam4e/gcc/gcc/startup_sam4e.o
  Building out/compile_time_request.o
Version: v0.7.0-457-g777013d-20190517_130421-ridingrasp
  Linking out/klipper.elf
  Creating bin file out/klipper.bin
  Flashing out/klipper.bin to /dev/serial/by-id/usb-03eb_6124-if00
Entering bootloader on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.2:1.0
Device reconnect on /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.2:1.0
lib/bossac/bin/bossac -U -p /dev/serial/by-path/platform-3f980000.usb-usb-0:1.1.2:1.0 -e -w out/klipper.bin -v

Erase flash

Done in 4.801 seconds
Write 18048 bytes to flash (36 pages)
[==============================] 100% (36/36 pages)
Done in 0.276 seconds
Verify 18048 bytes of flash
[==============================] 100% (36/36 pages)
Verify successful
Done in 0.314 seconds
pi@ridingrasp:~/klipper $ lsusb
Bus 001 Device 006: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@ridingrasp:~/klipper $ ls /dev/serial/by-id/*
ls: cannot access '/dev/serial/by-id/*': No such file or directory
pi@ridingrasp:~/klipper $ ls /dev/ser* -als
0 lrwxrwxrwx 1 root root 5 May 14 15:17 /dev/serial0 -> ttyS0
0 lrwxrwxrwx 1 root root 7 May 14 15:17 /dev/serial1 -> ttyAMA0
pi@ridingrasp:~/klipper $ ls /dev/ttyA*
/dev/ttyAMA0
pi@ridingrasp:~/klipper $ sudo service klipper start
pi@ridingrasp:~/klipper $ 

One observation, after running the Duet firmware load, the "diags" led on the board is not lit as the bossa command is using the "reboot to flash" option, whereas after loading the Klipper firmware, the diags light is still lit. I've tried using the Windows Bossa program to load the Klipper firmware, using the same options (lock, Erase All, Boot to flash) and it still leaves the Diags light lit.

KevinOConnor, 2019-05-21:
Okay, I'm guessing this issue is now resolved then.  The LEDs are controlled by software - if you wish to set them then identify their corresponding micro-controller pins and add `[output_pin]` config sections (see config/example-extras.cfg for details).  Alternatively, the LED pins can also be set during "make menuconfig" under the "low level" options menu.

-Kevin

KevinOConnor, 2019-05-21:
The output indicates that serial mode (instead of USB mode) was selected during "make menuconfig".  If you wish to use USB, then you must enable USB.

-Kevin

mweber99, 2019-05-21:
Ok, what do I enter as the hex value that it asks for when enabling USB in the make menuconfig? Sorry for the noob questions, I appreciate your help.

KevinOConnor, 2019-05-21:
Clear the config by running `cd ~/klipper/ ; rm .config` at the ssh prompt.  Then run "make menuconfig", select "SAM3/SAM4", then select "SAM4e8e".  Leave everything else at its default values.  Then flash the micro-controller and continue with the installation instructions from that point.

-Kevin

mweber99, 2019-05-21:
That's getting me closer, thank you . Enclosed is the latest log, its compiling but failing on connect.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3203126/klippy.log)

KevinOConnor, 2019-05-21:
Do you have the duex expansion board?  If not, you should remove all the config sections that reference the sx1509.

-Kevin

jakep82, 2019-05-21:
It would probably be beneficial to comment out the duex lines from the Duet2 example config.  They shouldn't be enabled by default.

mweber99, 2019-05-21:
I removed the duex and still get errors so I commented out all of the extra stuff that I do not have hooked up. Log enclosed. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/3204109/klippy.log)

KevinOConnor, 2019-05-21:
That error just indicates that the micro-controller is in an error state - issue a FIRMWARE_RESTART command in the OctoPrint terminal window to clear the error.

-Kevin

mweber99, 2019-05-22:
Thank you, it's working now. You can close this issue.

KevinOConnor, 2019-05-22:
@FHeilmann - looks like the sam4e8e code is crashing hard when the sx1509 is enabled but not present.  Any ideas?

Also, it looks like users are getting confused with config/generic-duet2.cfg - what do you think about moving that file to generic-duet2-duex.cfg and creating a simpler generic-duet2-eth.cfg for those users without the duex?  (Issue #1635 seems to be a similar issue.)

-Kevin

FHeilmann, 2019-05-23:
@KevinOConnor my best guess is that the i2c code doesn't catch a case where no device responds. At least all the cases I could think of should produce proper shutdowns or timeouts. I'll have to take a look. I'm fine with separating the example configs or commenting out stuff. I'll try to get to that in the coming days.

KevinOConnor, 2019-05-23:
@FHeilmann - I was able to track down the i2c reboot on my sam3x8e and I applied a fix (24b1638c).  It was an unhandled timeout in i2c_write().

-Kevin

closed 2019-06-02 by KevinOConnor