Klipper3d/klipper #1641: heater pwm_cycle_time longer than 0.3 feasable?

simonkuehling, 2019-05-16:
Hi Kevin,

I am testing a Klipper setup on a printer with mains-powered heated bed and chamber heater - both switched through zero-crossing solid state relais (~1000W each). These types of solid state relais can only switch at 30Hz with a 230V/60Hz supply. In order to keep a good resolution of 1/60th PWM duty cycle increments I'd like to set a pwm_cycle_time of 2.0 seconds for the PID loop.

Right now this is capped to a max value of 0.3 in the code. Would it be feasable to increase this to 2.0?
This would be plenty of resolution for these heater setups and also reduce the noise on the mains curcuit quite a bit.

KevinOConnor, 2019-05-16:
Changing the cycle time to something greater than 300ms would require changes to the code.  The default Klipper cycle time is 100ms (or 10hz) which seems to be in range of your 30hz capable relay, so I'm not sure why you'd want to mess with it, however.

-Kevin

simonkuehling, 2019-05-17:
Sorry, I should have been more precise in my description and had a mistake on the switching frequency - let me explain... 

First, the 30Hz are wrong, my mistake... its 120Hz or 60Hz depending on the SSR used...

Solid State Relais for AC mains power usually only switch at no load, which is at the zero-crossing points of the voltage sine wave. At 60Hz mains frequency that's every ~8.3ms. The turn-on and turn-off times may be as long as 8.3ms, respectively (the relais just waits for the next zero crossing before changing state). Some SSRs do even only switch after each full AC period (zero point after rising edge for example..), which halves the number of switching opportunities and doubles maximum turn-on/turn-off delays.
With a 0.1s PWM period that would only be 12 (or 6, depending on the SSR) switching opportunities per cycle. At the longest currently possible period of 0.3s it is a duty cycle resolution of still only 36 (or 18).

Another reason for longer PWM periods would be to reduce the noise frequency in the mains supply introduced by the switching of high-power resistive loads like my 1000W heaters. Switching Power Supplies that are fed from the same mains line tend to have trouble coping with AC voltage fluctuation at certain frequencies - and the thermal response of 1000W heating coils is more than long enough to be driven at 2s PWM cycle time.

My main question would be: Despite an obviously needed code change, are there any algorithmic reasons against increasing the number? Would it break anything else?

Thanks for all your time and effort you put into this project!

KevinOConnor, 2019-05-21:
I understand, but I don't see how a longer cycle time would improve performance.  Ultimately, the limitation is the 8.3ms resolution - a longer cycle time wont improve that resolution.

If you wish to switch at a lower frequency then you could try using the watermark control algorithm.

The sensor temperatures are sampled every 0.3s - it would not make sense to use a cycle time longer than the sensor sample time.  I don't know what would break if you did that.

-Kevin

simonkuehling, 2019-05-21:
The effective resolution of the PWM output at the heating element is determined by the number of possible "power levels" that can be achieved with a zero-switching relay. Taking a 8.3ms switching cycle of the relay as a best-case scenario, this would be `( cycle_time / 8.3ms ) = ( 300 / 8.3 ) = ~36`.

Heated beds or chamber heaters are usually quite a bit overpowered to decrease heat-up times and consequentially only run at ~30% power after heat up.
With 36 "power levels", a 1000W heating element can be switched in 28W increments between OFF and ON. Let's assume our 1000W heater needs 300W after heating up to maintain stable temperature. But we are only able to switch between 280W (10/36 cycles) and 308W (11/36 cycles) - which will from my experience be to coarse to maintain a stable temperature. And it makes PID tuning (also autotune) very hard.

With a `cycle_time` of 2 seconds the effective PWM resolution is highly increased (240 power levels or ~4W increments on a 1000W heater)

KevinOConnor, 2019-05-21:
You seem to be saying that the PID algorithm would be more stable if the pid loop time is larger.  This may be true (due to greater effective resolution of the PID output), but I don't think it's an obvious conclusion (because the greater pid loop time will result in a slower response).

In any case, if you (or someone else) wishes to implement this, test it, and demonstrate its superiority then that's great.

-Kevin

simonkuehling, 2019-05-23:
Hi Kevin, thanks for your thoughts on this - I very much appreciate your rational and cautious approach on code changes and feature developments!

When I get to making some experiments and my results show significant improvements I will get back on this topic at a later date.
I'll close this issue for now...

closed 2019-05-23 by simonkuehling

DroneMang, 2021-01-24:
I have a large AC heated bed with an SSR, and the PID control creates too much noise in my AC lines, most noticeable as flickering lights in the room, I tried setting the pwm_cycle_time to 0.3, but this did not help, I also have an AC line filter installed, and tried installing an 1F25 SSR Relay Filter, but this did not help the flickering lights.  The watermark control is much better for this.  So a longer cycle time could be good for large printers so that PID control can be used.

https://www.digikey.com/en/products/detail/sensata-crydom/1F25/254221?s=N4IgjCBcoLQBxVAYygMwIYBsDOBTANCAPZQDa4A7AEwgC6AvvYTZOWAGJUCsd9QA

simonkuehling, 2021-01-25:
@DroneMang, I guess you are using a zero-crossing SSR for switching your AC to the bed heater? In that case a lower switching frequency would usually only stretch the time between flickers but not avoid it. That is due to the nature of a zero-crossing switch that will always connect and disconnect full power (one period of AC).
It might be less annoying - as you said that the effective switching frequency of watermark control on your heater seems to be ok for you - but not necessarily the best cure against your symptoms (which is a high switching load on a mains circuit that has other loads like light bulbs attached).
Did you have a look into phase controlled SSRs (thyristor controllers)? Those will actually control the power delivery to your heater within each AC cycle and thus avoid any flicker at all.

May I ask what the power specification of your bed heater is?

Gilabite, 2021-01-30:
I have the light flickering problem too, 1300 watt 110v heater, no setting in klipper makes it totally go away, just tolerable. I rent so dont have alot of options for changing things. @simonkuehling are you referring to SCR output SSRs? And is it phase angle control vs zero crossing? Im trying to find ones that arent zero crossing but they all seem to mention it, I usually buy from mouser or digikey. Would you happen to know of a good brand/model or a link to ones like you are referring to? Would love for my lights to not flicker. Thank you!

simonkuehling, 2021-01-30:
Hi @Gilabite, if I’m not mistaken and as far as I can tell from googling it right now, „SCR output“ seems to be another name referring to the type of silicon switch that can be used to implement phase angle control.
Yes, it is phase angle control vs. zero crossing switching.
Phase angle control switches on after some proportional time during the AC half-period and switches off at the zero crossing, effectively reducing the AC voltage (which is a kind of crippled sine-wave now).
Zero crossing switches do only ever switch on or off at a zero crossing of the AC sine wave.

Regarding a recommendation, I do not have used a phase angle one myself yet, but for high power SSRs the products from Sensata-Crydom are a well known brand. Their „PMP Proportional Control SSR“ Series offer phase angle control for high loads like this. Not cheap, though...
https://www.digikey.de/en/product-highlight/c/crydom/pmp-proportional-control-ssr

If someone else can recommend other components in this category, I‘m all ears!

Also, it is hard to find info in the datasheets if they work with a PWM signal on the logic level side - you might need to add a RC-filter to smooth the PWM to an analog voltage first.

(another strategy could be to change all affected lightbulbs to LED, those tend to be much less sensible to these voltage drops in the mains circuit - looking at the price tag for the mentioned SSRs, that could be a cheaper option 😉)

andreostrovsky, 2021-05-16:
@simonkuehling @Gilabite I think I've found a working solution to our problem. Check out this Chinese module floating around the usual sources: https://www.aliexpress.com/item/32963090456.html
It should be a drop in replacement for the SSR. Just feed in klipper's PID PWM output and it should scale the output voltage. I would definitely attach some heatsinks onto the ICs though. Unfortunately it's only rated up to 3A/660W, but that's still better than what's practical with dc heaters.

If you really need more power and are willing to pay for it, there's this one from KRIDA: https://www.tindie.com/products/bugrovs2012/pwm-16a-3500w-triac-leading-edge-dimmer-50hz-60hz/

Of course being able to set the pwm cycle to 2s would be a much better solution. But given that it's dear leader Kevin we're talking about, that'll never happen.

astorun, 2021-08-30:
I really dont understand the reasoning of limiting this to 0.3 or not implementing another solution in klipper. Those flickering lights gives me terrible headaches. In RRF with Duet 3 board this is not a problem since it has a proper pwm frequency settings (M950 Q sub-command to set PWM frequency in Hz. Valid range: 0-65535) to match it perfectly with frequency of the lights. Sadly I am not a programmer to chase down a solution to fix this. After I switched over to the klipper it simply makes all the lights in my house to dance. It is the only thing that I dont like with klipper so far.