Klipper3d/klipper #1642: MCU 'mcu' shutdown: Timer too close

Jerem1313, 2019-05-16:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3187357/klippy.log)
bonjour, je n'arrive pas a finir une impression ,j ai une Under 3 CREALITY. marlin et repetier firmware marche très bien, avec klipper je reçois ce message
"15:17:28 MCU 'mcu' shutdown: Timer too close"
ce n'est pas Raspberry mais un pc avec VirtualBox 6.0.
merci d'avance
[klippy.log](https://github.com/KevinOConnor/klipper/files/3187380/klippy.log)

Jerem1313, 2019-05-16:
apres une autre impression ,et avoir changé le câble USB toujours pareil….
[klippy.log](https://github.com/KevinOConnor/klipper/files/3188229/klippy.log)

Jerem1313, 2019-05-16:
problème résolu !!! le port serial Virtual dans Windows 10 avait un baudrate de 9600 ,il change a l'installation de VirtualBox merci

KevinOConnor, 2019-05-25:
I'm going to close this issue as I'm guessing a solution was found for the original issue.

-Kevin

closed 2019-05-25 by KevinOConnor