Klipper3d/klipper #1643: false temperature reading from MAX31855 calculation routine

simonkuehling, 2019-05-16:
Hi @KevinOConnor and possibly also @cruwaller,

I am currently setting up Klipper with a couple of MAX31855 thermocouple chips as sensors for an extruder, a heated bed and a heated chamber. Doing so I get odd temperature readings that are about twice the actual temperature of the sensor:
```
temperature from dedicated thermocouple thermometer: 23°C
temperature reported from Klipper: 46°C
```
The configuration is working just as documented via software SPI like so:
```
[heater_bed]
heater_pin: ar41
sensor_type: MAX31855
sensor_pin: ar25
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
smooth_time: 0.01
pwm_cycle_time: 0.3
#pid_integral_max:
min_temp: 0
max_temp: 180
spi_speed: 4000000
spi_software_sclk_pin: ar10
spi_software_mosi_pin: ar14
spi_software_miso_pin: ar4
```
To debug what is happening I gathered some runtime data from within the `calc_temp()` function in `klippy/extras/spi_temperature.py`
```
MAX31855_SCALE = 18
MAX31855_MULT = 0.25

class MAX31855(SensorBase):
    def __init__(self, config):
        SensorBase.__init__(self, config, "MAX31855")
    def calc_temp(self, adc, fault):
        if adc & 0x1:
            self.fault("MAX31855 : Open Circuit")
        if adc & 0x2:
            self.fault("MAX31855 : Short to GND")
        if adc & 0x4:
            self.fault("MAX31855 : Short to Vcc")
        logging.info('********* adc:\t %s *************' % bin(adc))
        adc = adc >> MAX31855_SCALE
        logging.info('********* stripped:\t %s *************' % bin(adc))
        # Fix sign bit:
        if adc & 0x2000:
            adc = ((adc & 0x1FFF) + 1) * -1
        temp = MAX31855_MULT * adc
        logging.info('********* temp:\t %s *************' % temp)
        return temp
```
Here is one data point as a reference:
```
adc: 0b00000010111000000011001100000000
stripped: 0b10111000 (= decimal 184)
temp: 23
```
Looking into the MAX31855 datasheet the calculation procedure seems fine, basically reading bits 18..30 for the temperature and multiply by 0.25.
Further inspection suggested that the whole stream of bits is shifted to the left by one bit as it is provided to `calc_temp`. I double checked by physically disconnecting the thermocouple to introduce a known "Open Circuit" error state in the SPI readout from the MAX31855 chip - which confirmed the assumption.

I did not yet manage to find exactly where this happens in the SPI code, maybe you have an idea where this is introduced exactly?

A hacky workaround would be to bitshift the `adc` argument as the first line in `calc_temp()` - but that's not resolving the real bug I guess...

KevinOConnor, 2019-05-22:
Without the log there is little we can do.  It looks like there is a similar report at #1664.

-Kevin

simonkuehling, 2019-05-23:
Oh, beautiful, that's it! Problem already solved with your recent commit 6268c70 - it was just using the wrong SPI mode...

Thumbs up!
-Simon

closed 2019-05-23 by simonkuehling