Klipper3d/klipper #1644: Fresh installation, at least no temp reading

andrico21, 2019-05-16:
Had working printer with the same controller, now trying to setup new printer with the same controller.
Installed Klipper from scratch, pin configuration is the same as before on old printer. Flashed Due - no temp reading, only zeros on both sensors.
Ok, replaced RPi sdcard with old installation - got no temp reading also. Reflashed Due with previous Klipper installation from this sdcard - temp reading is ok, everything fine.
Suspect something was changed in code, need help. Tried both in serial and USB native mode.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3188830/klippy.log)

jakep82, 2019-05-16:
From your log: `NoOptionError: No option 'step_pin' in section: 'extruder'`

You didn't define step, direction, or enable pins for your extruder.  I'm guessing everything will work fine once you do that.

andrico21, 2019-05-16:
I'll check, not configured extruder yet because see no errors in terminal.

andrico21, 2019-05-16:
this one I got with old sdcard:
[klippy-old-working.log](https://github.com/KevinOConnor/klipper/files/3188918/klippy-old-working.log)

andrico21, 2019-05-16:
@jakep82 damn, you was right - the only issue was absense of [extruder] options. Thanks mate, now it works!

closed 2019-05-16 by andrico21