Klipper3d/klipper #1646: How to undo Enhanced delta calibration

emailcraigs, 2019-05-17:
How can I undo the changes made during Enhanced delta calibration? I like the results I was getting after doing just basic calibration better than after Enhanced delta calibration. Can I just delete the auto generated section from the config or do I need to also modify the running config somehow? Or do I have to start over....and if so what is the best way to do that with Klipper?

Hywelmartin, 2019-05-17:
remove from config.cfg and rerun the delta calibration works.. I don't know if there is another way..

emailcraigs, 2019-05-17:
So you are saying I can remove the auto-calibrated lines from the config file, but then I need to rerun the basic config steps? I guess what I am unsure about is how the Klipper program works. Does it run solely based on the config file or does it also have data that it has saved that I would need to expunge from it to revert?

Hywelmartin, 2019-05-18:
Klipper only saves data to the printer.cfg...

closed 2019-05-18 by emailcraigs