Klipper3d/klipper #1647: HE3D K280 huge delta config request

peterlpop, 2019-05-17:
Anyone here has a K280 config file that I could just copy and save me a day worth of work? It's a 280x600mm tall delta printer (single extruder, part fan, MKS L 1.0 ramps board, with the nozzle calibration thingy. 

If not, I guess I should post it after I get it all going.

peterlpop, 2019-05-17:
ok, so the Printer is working with the delta sample file. - had to reverse the motors and end stops and adjust dimensions, but the delta_calibrate did a good job of finding where the bed is :)

how do I flip the button rotation around on the 2004 LCD?

# "RepRapDiscount 2004 Smart Controller" type displays
[display]
lcd_type: hd44780
rs_pin: ar16
e_pin: ar17
d4_pin: ar23
d5_pin: ar25
d6_pin: ar27
d7_pin: ar29
encoder_pins: ^ar31, ^ar33
click_pin: ^!ar35

Thanks.

Hywelmartin, 2019-05-17:
encoder_pins: ^ar33, ^ar31

but then they are i reverse for other things...

KevinOConnor, 2019-05-25:
I'm going to close this issue as it appears a solution was found for the original question.

-Kevin

closed 2019-05-25 by KevinOConnor