Klipper3d/klipper #1648: Error Behavior in OctoPrint (Disconnect for Trivial Things)

VanKurt, 2019-05-17:
I recently updated my printer to Klipper and so far I'm loving it :-)
There is only one issue that keeps annoying me:

Whenever an "error" occurs, OctoPrint immediately disconnects from the printer. This might make sense in some cases, but with Klipper it even happens for simple, uncritical things: e.g. I want to start bed levelling, but forgot to home the printer first. In that case an error pops up and OctoPrint disconnects. This is just one example, but disconnects like that happen to me A LOT.
IMHO it would be enough if Klipper refused the bed level command and a popup would come up in OctoPrint. No need to shut it all down.

I'm not sure if this is a Klipper issue, a OctoPrint issue or if I'm just holding it wrong ;-)

theopensourcerer, 2019-05-17:
This sounds like you haven't configured Octoprint to re-connect/stay connected...

https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#configuring-octoprint-to-use-klipper

> In the Settings tab, navigate to the "Behavior" sub-tab and select the "Cancel any ongoing prints but stay connected to the printer" option. Click "Save".

VanKurt, 2019-05-17:
Ah, dang it! Should have read the instructions more carefully ;-)
Thank's a lot!

closed 2019-05-17 by VanKurt