Klipper3d/klipper #1649: BLTouch V3 timing imprecise

HorstBox84, 2019-05-17:
I attached a BLTouch V3 to an Ender 3 running Klipper and used the `BLTOUCH_DEBUG` commands.
The `self_test` and `pin_up` work just fine but `pin_down` doesn't work.

I measured the signal with an oscilloscope and `pin_down` is 700µs instead of 650µs (+-20µs). Afterwards I checked the bltouch.py and there the defines for the commands seem to be wrong.

```python
SIGNAL_PERIOD = 0.025600
MIN_CMD_TIME = 4 * SIGNAL_PERIOD
```

In my opinion `SIGNAL_PERIOD` should be 20ms and not 25.6ms. Too long a period shouldn't be a problem but you'll loose resolution.

```python
Commands = {
    None: 0.0,
    'pin_down': 0.000700,
    'touch_mode': 0.001200,
    'pin_up': 0.001500,
    'self_test': 0.001800,
    'reset': 0.002200,
}
```

`pin_down` needs to be 650µs (+-20µs) and not 700µs which is too far off. The same goes for `touch_mode` which should be 1165µs and `pin_up` which should be 1475µs. The other commands are inside the spec.

I am not sure if the PWM ressolution of the Ender3 board is sufficient to give a correct signal but the timing definitely needs to be improved as I tested my BLTouch V3 with an function generator and a correct 650µs signal and there the pin_down does work.

The older BLTouches seem to be more forgiving. With the V3.0 they added some new commands and I guess therefore the improved the accuracy.

If needed I can post some screenshots.

HorstBox84, 2019-05-17:
In the following picture you'll see the `pin_down` pulse when using the BLTouch functionality. It is definitely too long and isn't recognised by my BLTouch V3.0.
[image]

Here you'll see the `self_test` signal which is fine and working with my BLTouch.
[image]

Now I tried to use the old `[probe]` method to fine tune the pulse length via the servo angle. The next step below 700us seems to be 625us (I guess timer resolution is the issue here). This is recognized by my BLTouch but not always on the first try. Is there any possibilty to come closer to 650us?
[image]

sniperlucian, 2019-05-17:
@HorstBox84 

thank you very much for you findings. My clone (transparent) didn't  trigger pin_down 100%. Also it seems that close to 80°C heat bed the internal clock (RC based??) is running away, making it worse. 

Changed to your timings and will report back after testing.

KevinOConnor, 2019-05-20:
It looks like antclabs changed the hardware specifications on their website.  For reference, this document shows what the antclabs website used to recommend: https://cdn.shopify.com/s/files/1/0588/5745/files/BLTouch-Classic-7.pdf .  It is quite obnoxious of them to attempt to retroactively change the specifications.

It may not be straight forward to support a 40us pulse width window from Klipper.  It's unclear what changing the pulse width will do to the many bltouch clones.

-Kevin

HorstBox84, 2019-05-20:
@sniperlucian
You're welcome. It just drove me nuts that some commands worked and some not. I just had to investigate.
Even if you change the value you won't get pin_down to work 100% as it seems to be impossible to get the correct timing with the used timer resolution. With 625µs you will be still 5µs out of spec.

@KevinOConnor 
Ahhh that is where those numbers came from. I was wondering why nearly all values were off a bit. You are of course right about the Bltouch clones but as it is at the moment the genuine Bltouch V3 doesn't work or if it works it is by coincidence. Best way would be to make the Bltouch V3 work with the correct timing and implement a kind of legacy mode for the old/clone versions.

I had a short peek into the sources but didn't see what resolution the timer used for the pulses has.
Do you know the resolution of the timer?
Perhaps the resolution can be increased. I saw something about 8-/10-/12- and 16-bit timer modes somewhere in the Atmel specs.

KevinOConnor, 2019-05-20:
Given how badly antclabs has implemented the v3, I'm not sure it makes sense to consider it the "genuine" implementation.

Klipper uses software pwm for the bltouch (and servos in general).  As currently implemented, the limitation on resolution is 1/256th of the cycle time.  (This is why the cycle time is currently 25.6ms - as that allows us to request precise pulses in increments of 100us.)

-Kevin

sniperlucian, 2019-05-20:
I have two different clones,  from makerbase (transparent) and Geeetech. 

I had the impression that the makerbase ignored randomly the push_down event, since it didn't push down and also didn't show an error. After I changed the  timings (on AVR) it also didn't push down reliable, but always showed an error. Cleaning/adjusting didn't help. It's still doesn't come out every ~6 times - but works again sometimes after a reset event. 

So I switched to the geeetech. This one has a quite loose top screw, but after fixing that it works reliable. 

I found the same datasheet as Kevin, so I guess at the end the change of timing didn't to a thing to my clones though.

KevinOConnor, 2019-05-20:
FYI, I've attempted to update the BL-Touch timing on the bltouch-20190520 branch (PR #1661):

cd ~/klipper ; git fetch ; git checkout origin/bltouch-20190520 ; make ; sudo service klipper stop ; make flash ; sudo service klipper start

-Kevin

SimonOrdo, 2019-05-22:
> FYI, I've attempted to update the BL-Touch timing on the bltouch-20190520 branch (PR #1661):
> 
> cd ~/klipper ; git fetch ; git checkout origin/bltouch-20190520 ; make ; sudo service klipper stop ; make flash ; sudo service klipper start
> 
> -Kevin

@KevinOConnor 
I have a genuine V3 and this changelist works for me.  Everything is working now. Thanks!

HorstBox84, 2019-05-23:
@KevinOConnor: Thanks a lot for this quick fix.

@SimonOrdo: Thank you for testing.

I am printing 24/7 at the moment because of an upcoming event an can't measure it while printing.
I hope I'll find some time this weekend to measure the timing with the fixed branch. Then I'll post the pictures of the timing here.

HorstBox84, 2019-05-26:
As promised here the results:

Pin Down:
[image]

Self Test:
[image]

Works like a charm now. Thanks a lot!

KevinOConnor, 2019-05-26:
Thanks everyone.  I committed the changes to the master branch (commit 7702cc29).

-Kevin

closed 2019-06-02 by KevinOConnor