Klipper3d/klipper #1650: Question on usage (SKR v1.3, Ender 3)

Firedrops, 2019-05-17:
Thank you for publishing a generic profile for the SKR V1.3! Very helpful for complete newbies like me.

Please bear with me, I'm not familiar with what I'm supposed to do. I've worked out what I think should be done, could someone please confirm if they are right?

I am switching a stock Ender 3 Pro motherboard to an SKR V1.3 with TMC2208 v1.2 stepper drivers.
Are we supposed to un-comment all the lines in the
########################
TMC2208 configuration
########################

AND comment out the corresponding lines in
########################
Main configuration
########################
Basic steppers configuration
?

ALSO,
replace  
`sensor_type: ATC Semitec 104GT-2` and 
`control: watermark` in `generic-bigtreetech-skr-v1.3.cfg` 
with 
`sensor_type: EPCOS 100K B57560G104F` and 
`control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182`

and the whole of `[kinematics]` and `[display]` sections with the values in `printer-creality-ender3-2018.cfg`?

zwnk, 2019-05-17:
https://gist.github.com/zwnk/92066555145cddecfd7df92a713dc5f1

a working printer config for ender 3 with tmc2208 on x,y and z.

Firedrops, 2019-05-17:
Sweet! Thank you!

I noticed that your `[tmc2208 stepper_*]` configs are different from the ones in `generic-bigtreetech-skr-v1.3.cfg`. In that case, how should I implement a `[tmc2208 extruder]`?

And is there anything else I should be careful of/change?

zwnk, 2019-05-17:
the correct current values for an ender 3 are(they are motor specific):
x,y,z:
`run_current: 0.6`

extruder:
`run_current: 0.7`

Firedrops, 2019-05-17:
> the correct current values for an ender 3 are(they are motor specific):
> x,y,z:
> `run_current: 0.6`
> 
> extruder:
> `run_current: 0.7`

but I noticed that you have set `run_current: 0.7` for x, y, and z in your `printer.cfg`. Is that a typo that should be changed to `0.6`?

also, none of the motors have a `hold_current` set, unlike in the `generic-bigtreetech-skr-v1.3.cfg`, is that fine?

JamieAKing, 2019-05-18:
If the motors get hot the turn down the current but .1 amp isn’t going to do a lot. By not setting a hold current it will default to the same value as the run current. 
If a motor skips steps increase the current, if it gets hot lower the current, it is very machine specific. No two Chinese motors are the same, even belt tension can effect the current needed. Play with it, you can’t break it (don’t go silly and try setting max current as 2208 is a pokey little chip for these small motors)

povlhp, 2019-05-22:
I am having problems with tmc2208 as well. Bought bigtree with the TMCs. I have tried:
```
[tmc2208 stepper_x]
uart_pin: P1.17
microsteps: 16
interpolate: True
run_current: 0.7
stealthchop_threshold: 200
driver_PWM_AUTOGRAD: True
driver_PWM_AUTOSCALE: True
driver_HEND: 3
driver_HSTRT: 0
#tx_pin: P4.29
#microsteps: 16
#run_current: 0.800
#hold_current: 0.500
#stealthchop_threshold: 250
```

Kept the UART pin and tried the commented lines as well. With and without tx_pin
Without this stuff, steppers works. 

With any of the 2 sets above I get:
Unable to read tmc2208 'stepper_x' register IFCNT

Here is the part from the log:
```
Loaded MCU 'mcu' 74 commands (v0.7.0-469-g6268c70-20190522_200625-orangepilite / gcc: (15:5.4.1+svn241155-1) 5.4.1 20160
919 binutils: (2.28-5+9+b3) 2.28)
MCU 'mcu' config: BUS_PINS_i2c1=P0.1,P0.0 SOFT_PWM_MAX=256 STEP_DELAY=2 ADC_MAX=4095 BUS_PINS_ssp0=P0.17,P0.18,P0.15 RES
ERVE_PINS_USB=P0.30,P0.29,P2.9 CLOCK_FREQ=100000000 MCU=lpc176x STATS_SUMSQ_BASE=256
Sending MCU 'mcu' printer configuration...
got {'#receive_time': 1554.92906549, u'crc': 2184021049L, u'is_shutdown': 0, '#name': u'config', u'is_config': 1, '#sent
_time': 1554.927609782, u'move_count': 1024}
Configured MCU 'mcu' (1024 moves)
Config error
Traceback (most recent call last):
  File "/home/octoprint/klipper/klippy/klippy.py", line 137, in _connect
    cb()
  File "/home/octoprint/klipper/klippy/extras/tmc2208.py", line 327, in _init_registers
    self.set_register(reg_name, val)
  File "/home/octoprint/klipper/klippy/extras/tmc2208.py", line 348, in set_register
    self.ifcnt = ifcnt = self.get_register("IFCNT")
  File "/home/octoprint/klipper/klippy/extras/tmc2208.py", line 340, in get_register
    "Unable to read tmc2208 '%s' register %s" % (self.name, reg_name))
Error: Unable to read tmc2208 'stepper_x' register IFCNT
Unable to read tmc2208 'stepper_x' register IFCNT
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready
```

zwnk, 2019-05-22:
depending on your tmc2208 you need to solder a connection between two 2 or 3 solderpads on the bottom side and/or connect the uart pin to the 4 NC pin.

povlhp, 2019-05-22:
Go the BigTreeTech modules. 1 pads. Center and J1/J2.
I can see I can bridge all 3 - to connect to both NC and PDN.
But looking at official docs, seems like bridging J1 and center is the way to go to get UART:
https://github.com/bigtreetech/BIGTREETECH-TMC2208-V2.1/blob/master/UART.jpg

And I figured out, the suspected dead ender-3 stepper was not that, but "only" a bad stepper motor cable.

But now I am doing the upgrade all the way. Default I got twice the stepsize upgrading (before setting microsteps). So when going to right of center (for bltouch homing), my x hit the right side. Another lesson learned. 

Any reason not to go to 128 microsteps ? At least on extruder ? For this board, steprate should not be a problem.

povlhp, 2019-05-23:
An update to myself. I needed to solder J2 despite the J1 being clearly listed on the BigTreeTech github. So I bridged all 3 pads. And now it is working for me. There is an issue on this at bigtreetech github for the TMC2208 v2.1 whcih came bundled with my board.

KevinOConnor, 2019-06-02:
I'm going to close this issue as it looks like a solution was found for the original question (skr config file for the ender3).

-Kevin

closed 2019-06-02 by KevinOConnor