Klipper3d/klipper #1651: BLTouch Crashing Into Bed

JBrew13, 2019-05-17:
Hey everyone,

I'm pretty new to the whole 3D Printing biz, but I have an Ender 3, which I've installed Klipper on. So far, its worked fantastic. I've recently purchased a BLTouch (authentic from ANTCLAD) and did the wiring and install. The BLTouch responds to commands from OctoPrint (pin up, down, reset, etc) but when I go to home the axis and then do a PROBE_CALIBRATE, the BLTouch crashes into the bed and goes into a flashing red error status. 

The wiring is correct, and I've done every variation on the 
pin_up_touch_mode_reports_triggered: True/False
pin_up_reports_not_triggered: True/False

with no luck. I've altered the pin move time, z_offset...nothing seems to work. 

attached are the klippy and printer config files. GitHub doesnt do .cfg apparently, so i put that into a text file. Any insight/ideas would be greatly appreciated.

[printerconfig.txt](https://github.com/KevinOConnor/klipper/files/3192017/printerconfig.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/3192006/klippy.log)

xelprep, 2019-05-19:
Did you do the thumbnail test to see if the virtual endstop is working? While homing Z, try to gently trigger the BLTouch with the flat part of your thumbnail before it comes close to the bed and see if Z homing stops immediately, or at all, and cut power before it crashes into the bed. I had to [remove the c7 capacitor](https://www.antclabs.com/wiring3) on my Ender 5 board, which is the same board as the Ender 3, for this to work correctly. 

Make sure your z-offset is high enough to give you some wiggle room when you calibrate it with PROBE_CALIBRATE. I used 5mm initially for safety.

I also have a genuine BLTouch V3 which you most likely also have. I had to use `pin_up_touch_mode_reports_triggered: False` in my config for it to work properly, even though that setting is described as being needed just for clones.

JBrew13, 2019-05-20:
After the first time it crashed into the bed, i used one of the XYZ cubes as a standoff so that it wouldnt crash into the bed, but no that doesnt stop it either. Everything else works, I can completely control and reset from OctoPrint, it just doesnt work like it should with the virtual endstop. I tried doing the pin_up_touch_mode_reports_triggered: False as well as the other option and every combination of such with no avail. I'm definitely hesitant to remove the cap, for obvious reasons, but if thats what worked for you then i might have to give it a go. Is this a straight up surface mount cap or can it just be removed without de-soldering? Lastly, does this affect the actual z-endstop at all if I decide to just remove the BLTouch and go back to normal operation?

xelprep, 2019-05-20:
Everything I've read and experienced indicates this capacitor never made sense to have installed there and even the stock z-stop would benefit from its removal. I found some good analysis somewhere on reddit regarding this capacitor on these boards, so feel free to do some googling to build your confidence. Honestly, I was pretty terrified to pull it off, but it was a very easy process. I did not desolder it, I just grabbed it with a tiny set of pliers and then ensured there was no material remaining between its solder points. It came off very easily.

JBrew13, 2019-05-20:
Yeah, I was just doing some reading on the webpage you linked in the previous comment, seems pretty straightforward. I'll give it a go tomorrow and see what happens. 

Seems like a lot of people are having a rough go with the Ender 3 and the BLTouch with Klipper, so if this works, it might mitigate the point for lots of folks. Thanks again for your help, I'll comment back once the deed is done :)

JBrew13, 2019-05-21:
and its official, removing the capacitor did the trick...much obliged!

closed 2019-05-21 by JBrew13