Klipper3d/klipper #1653: Klipper printing slower

bobathefetts, 2019-05-18:
I was running unified firmware on my printer before upgrading to klipper. After I installed klipper my printer is printing a lot slower. How do I speed up the printing?

bobathefetts, 2019-05-18:
my raspberry pi doesn't have a /tmp/klippy.log file

timmit99, 2019-05-18:
Did you issue an M112 command to generate the log file?
What software are you using to view the Pi's directories?

bobathefetts, 2019-05-18:
yes i did issue that command i just wasnt looking in the right folders. attached is the file
[klippy.log](https://github.com/KevinOConnor/klipper/files/3194317/klippy.log)

mwr666, 2019-05-19:
Hello,
I think You have default setting for square_corner_velocity parameter. By default it's set to 5mm/s. My system run well at 20mm/s square corner vlocity. You need to find Your own speed for this parameter. Try adding this line: `square_corner_velocity: 20.0` so Your config at [printer] settings will look like this:
`[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100
square_corner_velocity: 20.0`

Hywelmartin, 2019-05-19:
Is it the z max speed that interfere?
Try to print without mesh bed leveling

KevinOConnor, 2019-05-21:
Nothing in the log looks abnormal.  Normally the way to increase print speed is by setting a greater velocity when slicing the desired print.

-Kevin

closed 2019-06-09 by KevinOConnor