Klipper3d/klipper #1654: Do you agree with Simplify3d's statement below?

w1ebr, 2019-05-18:
Hi Kevin,

I am questioning the support team at Simplify3d on some of the tool paths their software is using. In the process of this I sent them a small sample of gcode and attached the travel path their previewer shows which I claimed is incorrect. I have attached the "evidence" below.

Simplify3d's response was this: 
> Thanks for your email. However, the preview mode or for that matter G-code in absolute mode does not work as indicated in the document. The movement is always based on the last location, irrespective of the active tool. This is indeed correct as it is how firmwares interpret the commands as well. G-codes are not associated with a particular toolhead.

Is this also your understanding?

Gene

[testgcode.pdf](https://github.com/KevinOConnor/klipper/files/3194144/testgcode.pdf)

JamieAKing, 2019-05-18:
the Path drawn is correct but the explanation is wrong, absolute (G90) moves to a give co-ordinate, incremental (G91) moves relative to the current co-ordinate 
ie
G90.                   -set absolute mode
G1 x100 y100    - moves to x100 y100 regardless of current position 
G91.                    -set incremental mode 
G1 X10 y10.      - moves to x110 y110

w1ebr, 2019-05-18:
The gcode I showed mentions neither G90 or G91 in it. My question is whether T0 and T1 have unique coordinates in the firmware, or, as Simplify3d said _"G-codes are not associated with a particular toolhead."_

w1ebr, 2019-05-19:
Printer firmware has to be configured properly to allow for independent carriages, or fixed offsets, for multiple extruders. Simplify3d can handle simple x and Y offsets between extruders on the same carriage. It has an option on the GCODE tab for "Use independent extruder axes" but says not to use for Marlin type firmware because Marlin doesn't support it ( I know Marlin supports IDEX machines and the current location changes when the active tool changes). I want to keep the dragging of an extruder over an area being printed by the other extruder but with Simlify3d, it seems that every tool change means that it does not know where the new carriage is initially located.

JamieAKing, 2019-05-19:
Above my pay grade I’m afraid, I have only used single carriage machines,

megablue, 2019-05-20:
> Printer firmware has to be configured properly to allow for independent carriages, or fixed offsets, for multiple extruders. Simplify3d can handle simple x and Y offsets between extruders on the same carriage. It has an option on the GCODE tab for "Use independent extruder axes" but says not to use for Marlin type firmware because Marlin doesn't support it ( I know Marlin supports IDEX machines and the current location changes when the active tool changes). I want to keep the dragging of an extruder over an area being printed by the other extruder to a minimum but with Simplify3d, it seems with every tool change Simplify3d does not know where the new carriage is initially located.

use z-hop on every tool change. for s3d specifically it is not treating a multi mesh model as a whole, hence even if the second tool shared the exact same coordinates as the previous toolhead, it is treating the meshes as individual print per toolhead and cannot avoid traveling above the printed parts of other toolheads. also,most slicers do not keep track of where the toolheads are/were except for the current toolhead, in most situations you had to be clever with the scripting to either use relative coordinates or a fixed absolute coordinates to park the inactive toolheads smartly/hackishly.

KevinOConnor, 2019-05-21:
> The gcode I showed mentions neither G90 or G91 in it. My question is whether T0 and T1 have unique coordinates in the firmware, or, as Simplify3d said "G-codes are not associated with a particular toolhead."

Klipper does not store a gcode toolhead location for each extruder.  As far as I know, Klipper is similar to other common firmware in this regard.

-Kevin

w1ebr, 2019-05-21:
Then how do you know how to move the carriage to the next location if the software doesn't know its' current location? I have an IDEX setup and I think the software has to know what the carriage coordinate is. When using Marlin previously, when I changed active tool the coordinates displayed would change to show the carriage's current position.

KevinOConnor, 2019-05-21:
On a "dual carriage" system, Klipper does track the last commanded position of each stepper motor.  On a SET_DUAL_CARRIAGE command, it will translate the stepper motor positions into a g-code position and set the current g-code position to that.  There isn't two g-code positions though - once the new g-code position is set, the previous g-code position is discarded.

Traditional multi-extruder printers (where multiple extruders are on the same carriage) just update the toolhead g-code position via the SET_GCODE_OFFSET command.

-Kevin

closed 2019-06-09 by KevinOConnor