Klipper3d/klipper #1655: Can't home corexy on new Fysetc F6

FettFanatic, 2019-05-18:
I am trying to set up my new Fysetc F6 board on my corexy machine.  I am unable to get my machine to home correctly.  I previously had a MKS Gen 1.4 using Klipper and all motion control was working ok (temperature control wasn't working with that board which is why I bought a new board).  I home to the front left of my machine.  When I home the carriage moves in a diagonal direction.  The Y slider moves toward the end stop but the carriage moves away from end stops.  If I manually trigger the end stop it goes to home the next axis and moves the carriage towards the end stops but the y slider moves away.  

When I did the stepper buzz it only moved just the x motor or just the y motor and not both as would be needed to get x and y motion on a corexy system.  I didn't know if that was expected behavior or not.  I have tried inverting direction pins.  I have pulled the latest version of Klipper.  I deleted the printer.cfg and made a new one just in case it had somehow gotten corrupted.  I am at a loss for what else to try.

Any help would be appreciated.  Thanks.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3194155/klippy.log)

JamieAKing, 2019-05-18:
Stepper buzz should only move one motor either the motor configured as stepper_x or stepper_y. Can you verify that they both pulse independently when buzzed? This would rule out a Wiring issue which is usually the culprit of diagonal movements, either that or a driver not installed correctly. 
I had no trouble with motion from the f6 on my core xy other than the tmc2208’s but I see you are not using them so that is a rabbit hole avoided

JamieAKing, 2019-05-18:
I’m no expert but addding this to your config file will allow you to set a zero without homing and then move in just x or y, may help diagnose the problem 
#[force_move]
#enable_force_move: False
#   Set to true to enable FORCE_MOVE and SET_KINEMATIC_POSITION
#   extended G-Code commands. The default is false.

FettFanatic, 2019-05-19:
Yes.  When I buzz x only x moves and when I buzz y only y moves.  I am using Panucatt SD6128 for my stepper drivers.  I pulled them and put them back in just to make sure it wasn't a bad connection.  I am using 32 microstepping with jumpers in the first and third designated slots with the one other jumper it shows to put in for steppers other than TMC.

I stopped the homing about mid axis just to see what would happen if I did a manual move after that.  X and Y moves were both diagonal if I called for positive or negative moves.

Arksine, 2019-05-19:
It sounds like one of your dir pins needs to be inverted.  When you buzz does each stepper begin moving in the same direction?

FettFanatic, 2019-05-19:
I have inverted both axis one at a time and I also inverted both axis together just to verify and nothing really changes except for the direction of the diagonal move.

JamieAKing, 2019-05-19:
When it makes the diagonal move are both motors turning? Can you make the diagonal moves in all directions? Does the extruder operate as expected

FettFanatic, 2019-05-19:
Yes both motors are turning when I try to home and also when I call for a manual move through octoprint.

It does the diagonal move with a G28 X or a G28 Y and both motors move then too.

The extruder and z both operate normally.

I can't move the carriage to the front left or to the back right.  It always moves in a diagonal between front right and back left.  The carriage will move either direction while making that diagonal move but again never towards the front left or back right.

I switched the drivers into the extruder 1 and extruder 2 slots and changed the pins in the printer.cfg just to see if I had a bad driver socket or bad board in some other way and it is still the same.

I am going crazy trying to get this to work.  It has been so long since I have been able to print anything I am having withdrawals and my wife want the kitchen table back.  :)

JamieAKing, 2019-05-19:
Sorry for all the questions but hard to diagnose without seeing it. I think your definitely fighting a hardware issue as apposed to a klipper issue. My f6 worked great out of the box with klipper (2208 aside) i had trouble getting things moving in the right direction (reversed motor connection/homing direction) but I always had linear movement, is it possible one of the motors is skipping and not physically turning. This would be indicative of a low current setting on one of the drivers. Diagonal movement on core xy requires one motor to turn and the other remain stationary. The fact that diagonal movement can only be achieved along one vector also points to a non moving motor. If it’s not the current power off 5e board and double check the jumpers is my best suggestion. There’s a lot of them on the f6 and it’s easy to miss a leg or hit the wrong post. There are also jumpers above the limit switches used for switchless homing on the tmc drivers which need to be removed (thou I doubt that is the problem if z is working)

FettFanatic, 2019-05-19:
Both stepper drivers read 0.92 when I check their voltage.  On my drivers that would be 1.8 amps to the motors which is within the range for both the driver and motor.  I switched the x and y motor plugs just to check and it now just does diagonal movements from front left to rear left.  It is just the same diagonal in a different direction.  I removed all the jumpers above the limit switches except the one below the z that feeds it 5v.  When looking at the board I have jumpers on the first, second and fourth set of pins on the row right below the cap for 32x microstepping.  I am at a loss.

FettFanatic, 2019-05-19:
Well I am now able to home correctly.  

It turned out to be a bad stepper driver.  I had literally just pulled it out of a board where it was working and didn't even think that could be the issue.  

Thanks for the help.

closed 2019-05-19 by FettFanatic