Klipper3d/klipper #1657: Compiling for LPC1768 fails?

rlesac, 2019-05-19:
Using BananaPi.
Can compile for AVR without issues (was using Klipper on a Melzi board), but fails when I swithc to LPC1768.
Any suggestions?

`
bananapi:~/klipper$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.2 LTS
Release:        18.04
Codename:       bionic

bananapi:~/klipper$ make menuconfig
bananapi:~/klipper$ make clean
bananapi:~/klipper$ make
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/spi_software.o
  Compiling out/src/lpc176x/main.o
  Compiling out/src/lpc176x/gpio.o
  Compiling out/src/lpc176x/adc.o
  Compiling out/src/lpc176x/i2c.o
  Compiling out/src/lpc176x/spi.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/alloc.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/../lib/lpc176x/device/system_LPC17xx.o
  Compiling out/src/lpc176x/usbserial.o
  Compiling out/src/generic/usb_cdc.o
  Building out/compile_time_request.o
Version: v0.7.0-457-g777013d-20190519_162058-bananapi
  Assembling out/src/../lib/lpc176x/device/TOOLCHAIN_GCC_ARM/startup_LPC17xx.o
  Preprocessing out/LPC1768.ld
  Linking out/klipper.elf

ERROR: The compiler failed to correctly compile Klipper
It will be necessary to upgrade the compiler
See: https://bugs.launchpad.net/ubuntu/+source/newlib/+bug/1767223

Makefile:89: recipe for target 'out/klipper.elf' failed
make: *** [out/klipper.elf] Error 99
`

KevinOConnor, 2019-05-21:
The problem (and resolution) is exactly as the error message above states.

-Kevin

sergeysi779, 2019-05-22:
I've recently encountered the same problem. Managed to manually downgrade (I think it's possible to upgrade too but downgrading was easier) the compiler and it compiled. Also you can compile MCU firmware on other OS, eg more recent Ubuntu.

closed 2019-05-26 by klipper-gitissuebot